jneilliii/OctoPrint-BambuPrinter #7: [BUG] 0.0.14 broke P1 cache filesize and date

jneilliii, 2024-02-18:
https://github.com/jneilliii/OctoPrint-BambuPrinter/commit/3889efa67a0fe4523333c57ce82197d7a25dca8d#r138761440

Pavulon87, 2024-02-18:
In my case (P1S) `ftp.list_files("cache/", ".3mf")` return only filenames without directory

> filelistcache = ['xx.3mf', 'BambuLab AMS Dryer pod with indent for AMS fix .3mf', 'Rzutki_Uchwyt.3mf', 'Bambu Bed Scraper.3mf', 'xxx_.3mf', 'ninjago_.3mf', 'xxx.3mf', 'tank_7e.3mf', 'Cube.3mf', '608 Bearing.3mf', 'Bag Clip - print in-place.3mf']

It is not necessary to remove "cache/" from the filename:
`filename = entry.replace("cache/", "")`
but it is required to add "cache/" when checking the file size:
`ftp.ftps_session.size("cache/"+entry)`

Apparently, this is handled differently in X1 and P1.

jneilliii, 2024-02-18:
Yep, definitely a difference in ftps implementation then. This is what I get with the one file I have uploaded.
```
['cache/LCAC_large_PLA_4h14m.3mf']
```
so easy enough to fix maybe replacing `{entry}` with `cache/{filename}` in subsequent calls for filesize and date.

jneilliii, 2024-02-18:
added in above commit and released as version 0.0.15. still works for me, please give it a try and report back.

Pavulon87, 2024-02-18:
> so easy enough to fix maybe replacing `{entry}` with `cache/{filename}` in subsequent calls for filesize and date.
Looks ok to me. I don't see any problems at the moment.

closed 2024-02-18 by jneilliii