jneilliii/OctoPrint-BambuPrinter #8: Connection Failure with Octopi and X1 Carbon.

jmf5123, 2024-02-29:
Seem to have no issues with install, but cannot complete or maintain a connection. Logs attached. 
[octoprint (2).log](https://github.com/jneilliii/OctoPrint-BambuPrinter/files/14453888/octoprint.2.log)

jneilliii, 2024-02-29:
which device? I've heard that the P1 and A1 devices can have this happen when other clients connect to it (ie slicer devices tab).

jmf5123, 2024-02-29:
Firmware is 01.07.02.00 (Latest). I am connected through the cloud. Should I move to LAN only? Only other I have connected is my  Bambu Studio app on my desktop.

jneilliii, 2024-02-29:
It might help figure out the issue if you enable debug logging for all the "bambu" options in OctoPrint's logging section. And then we can possibly get more information back on the communication side. I think it might be timing out on retrieving the file list possibly.

jneilliii, 2024-02-29:
> Firmware is 01.07.02.00 (Latest). I am connected through the cloud. Should I move to LAN only?

So I'm on an older firmware version on mine, running in local only mode.

jneilliii, 2024-02-29:
Yep, just checked the firmware via HA. I'm currently running 01.07.00.00 firmware version.

jmf5123, 2024-02-29:
I just went to LAN only mode and restarted Octoprint. Same issue. I even tried to start a print from the X1C to see if that would establish a connection. No luck. Maybe a firmware limitation?

jneilliii, 2024-02-29:
Would have to check with some of the other guys that are integrating to validate that. Just not sure if they locked down the connections or not in that version. I know they locked down rooting the firmware with that version and they may have added other stuff too.

jneilliii, 2024-02-29:
curious if you've tried power cycling the printer. still curious if it's dying on retrieving the file list or not.

jmf5123, 2024-03-01:
Power cycled with no luck. Must be something with the new FW. Was looking to use it mainly for secondary camera timelapse. Had it working with Home Assistant but only captured ~85% of the layers. Might go back to that.

jneilliii, 2024-03-01:
Weird, I'm using the same underlying module that the ha integration is using. Would still be helpful to have some debug logs.

jmf5123, 2024-03-01:
I reimaged the RPi to go back to HA, but what is weird that I can't connect to the X1C through HA using Lan Only. Had to go back to Cloud. I wonder with the latest FW the connection must occur through the cloud?

atax112, 2024-03-01:
Hey guys.

I mentioned this on the hass integration repo as well.

I started off online with my A1, home assistant out of the box, registering
with bambu etc.

Than I went lan only, still works, no issue. Hass integration, octoprint,
sending,  printing, orca running too, all at the same time.

S pozdravom / Üdvözlettel / With kind regards

Attila Tamas
***@***.***
0905 469 420

Clasko, 2024-03-01:
I can confirm this issue. X1C on 01.07.02.00

jneilliii, 2024-03-01:
Can one of you please enable debug logging for all the bambu items in OctoPrint's logging section and try the connection again, then after it fails share a system info bundle.

Clasko, 2024-03-01:
Here you go.
[octoprint-systeminfo-20240301140855.zip](https://github.com/jneilliii/OctoPrint-BambuPrinter/files/14462097/octoprint-systeminfo-20240301140855.zip)

As an aside. Connections via MQTTX (MQTT Client) are also extremely faulty and I get timeouts almost every time. But sometimes it does work. It seems as if very high timeouts have to be set here.

jneilliii, 2024-03-01:
Yeah, both your logs are showing roughly the same thing. A connection between the plugin and the printer is never actually created.

> 2024-02-29 16:30:12,331 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - bambu connection status: False

> 2024-03-01 14:07:18,287 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - bambu connection status: False

which is coming right after the connection attempt from code: 

https://github.com/jneilliii/OctoPrint-BambuPrinter/blob/616fdf7a825378c31d057020b8403346d292871c/octoprint_bambu_printer/virtual.py#L224-L225

Not sure if I can do anything about that from a local only perspective, but it may be possible to establish a cloud connection with Bambu's server, assuming that works with HA's pybambu module. Is the HA integration working properly for you @jmf5123 with the cloud option? 

The only thing I'm concerned about is if the local only connections are blocked, the ftps connection might not work to retrieve file listing information. Would one of you be able to test that with filezilla client for me?

Clasko, 2024-03-01:
FTPS is also not possible for me.

jneilliii, 2024-03-01:
So after some discussion with @WolfWithSword, one of the more experienced integrators of Bambu communications, the firmware version shouldn't be an issue. Since ftps is not possible it's leaning toward a potential network issue. He mentioned he's only seen it with specific configurations/equipment in the past.

_Do either of you have a non-standard networking setup? Such as multiple VLANs or unifi/ubiquiti equipment? Bambu connections can be spotty for both mqtt and ftps when going across vlans and sometimes don't play well with ubiquiti stuff._

jmf5123, 2024-03-01:
No non-standard set-up for me here. Moved my X1C back to the cloud and set up HA  using HAOS on an RPi4. No issues with HACS or the Bambu Lab Integration. Was able to set up controls and cameras with automation.

Clasko, 2024-03-01:
Yes, my Octoprint was in a separate VLAN. I moved it to the same VLAN as the X1C, but the problem is still not solved.
There is still a Ubiquiti switch involved. I will try to change this one as well. Thanks for the help.

jneilliii, 2024-03-01:
So the plugin should work in cloud mode, I just haven't added the settings to the UI to configure it and never tested. I'll try this weekend to go through the process and see if I can get it connected that way. Right now it would be a matter of manually editing config.yaml of OctoPrint to set these values.

```
plugins:
  bambuprinter:
    local_mqtt: false
    region: US
    email: <your bambu registered email address>
    auth_token: <not sure how to get this yet, but suspect it's similar to HA>
```

jneilliii, 2024-03-03:
I've just added the options for connecting to cloud and have done some initial tests. For everything to work properly, you need to have `cache cloud print files on micro sd card` option enabled on the X1 print options settings. After updating go into settings and uncheck `Use Local Access`, fill in registered email address and password and click login. if all goes well the Auth Token will be filled in. After that click save and then reconnect to BAMBU from the connection tab.

DmitryBellka, 2024-04-12:
Dear jneilliii
I tried to connect my X1CC of the Chinese region with the X1Plus firmware (based on 01.07.02), and it worked! This is amazing! Thank you very much!

Where can I read how to connect the built-in camera to the octoprint? initially, just a black screen

jneilliii, 2024-04-12:
Answered your question on the community forum. 

https://community.octoprint.org/t/octoprint-for-bambu-lab-a1-support-available/56294/55?u=jneilliii