jneilliii/OctoPrint-BambuPrinter #10: Installation instructions need fleshing out and minimum system requirements adding.

Darthmonkey, 2024-03-02:
Although the current installation instructions have comments stating they need updating, a little more instruction would allow people to start using and helping develop the software.
1. The instruction that the installation can be installed via the Plugin Manager is not correct, it's not listed in Plugin Manager when searching variations of 'Bambu', or 'octoprint-bambuprinter'.
2. System requirements need to be altered to comply with this installation, I'm getting:
 `
ERROR: Could not find a version that satisfies the requirement pybambu>=1.0.1 (from octoprint-bambuprinter)
ERROR: No matching distribution found for pybambu>=1.0.1
Looks like the plugin was already installed. Forcing a reinstall.
/home/pi/oprint/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmpwic34jyp/OctoPrint-BambuPrinter-master.zip --no-cache-dir --ignore-installed --force-reinstall --no-deps
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /tmp/tmpwic34jyp/OctoPrint-BambuPrinter-master.zip
ERROR: Package 'octoprint-bambuprinter' requires a different Python: 3.7.3 not in '<4,>=3.9'
Error!
Unexpected error while trying to install plugin from https://github.com/jneilliii/OctoPrint-BambuPrinter/archive/master.zip
`
If I install an earlier version of Octoprint, I may circumvent the second issue, but I'll have to do that and let you know.

jneilliii, 2024-03-02:
Yeah, I usually don't fill out those details until after I've finished development. The issue you're running into is the underlying python version. In order to get python 3.9 you'll need to reflash a pi with octopi 1.0.0. Older OctoPi 0.18 won't work because of the upstream python modules dependency I'm using (pybambu) requires 3.9. You can create a backup in your current setup, download zip, and use that zip after reflashing during initial setup wizard.

jneilliii, 2024-03-02:
readme updated at least for system requirements to make this clearer, and reference to plugin manager removed.

closed 2024-03-02 by jneilliii

Darthmonkey, 2024-03-14:
As a test, I've installed OctoPrint v 1.4.0 on a Ubuntu VM, which has **Python 3.10.12** running. This time I'm getting the errors:

> 2024-03-14 13:53:21,358   /snap/octoprint-pfs/56/usr/bin/python3 -m pip --disable-pip-version-check install https://github.com/jneilliii/OctoPrint-BambuPrinter/archive/master.zip --no-cache-dir
> 2024-03-14 13:53:23,845 > Collecting https://github.com/jneilliii/OctoPrint-BambuPrinter/archive/master.zip
> 2024-03-14 13:53:29,834 > Downloading https://github.com/jneilliii/OctoPrint-BambuPrinter/archive/master.zip
> 2024-03-14 13:53:30,921 ! ERROR: Package 'OctoPrint-BambuPrinter' requires a different **Python: 3.6.9 not in '>=3.9,<4'**

Should I assume I've gone too far ahead now?

jneilliii, 2024-03-14:
No, that version you installed it quite old and seems that it is using Python 3.6.9 not 3.10.12 as you listed. The snap version of OctoPrint is quite broken and not officially maintained by the OctoPrint team. If you want to get support I recommend using the octoprint_deploy script linked on the download page instead and that will give you a better starting point using newer Python version.