jneilliii/OctoPrint-BambuPrinter #9: need to disconnect and reconnect before each print start

hashoom8, 2024-03-02:
I'm printing through bambu studio 

but I need to disconnect and reconnect before each print start because octoprint did not detect the print started

jneilliii, 2024-03-02:
on the terminal tab do you see lines like `{filename} open failed` repeating?

hashoom8, 2024-03-02:
> on the terminal tab do you see lines like `{filename} open failed` repeating?

yes 
[image]

jneilliii, 2024-03-02:
I think I've fixed that in the last release. it should reload the filelist if the file isn't found in the file list. give version 0.0.16 a try and let me know.

hashoom8, 2024-03-03:
> I think I've fixed that in the last release. it should reload the filelist if the file isn't found in the file list. give version 0.0.16 a try and let me know.

will i did the update turned on the Printer and start the connection , start a print through bambu studio , and its not showing that I'm printing , i had to disconnect and reconnect 

[image]

jneilliii, 2024-03-03:
please enable debug logging for the bambu related entries in OctoPrint's logging section. and try the process again and then share your octoprint.log.

hashoom8, 2024-04-29:
> please enable debug logging for the bambu related entries in OctoPrint's logging section. and try the process again and then share your octoprint.log.

i found out what happening 

when i start a print from the slicer , octoprint can't find the file in the sd card , so when i do reconnect or refresh sd card files octoprint  detect it and showing print started

jneilliii, 2024-04-29:
Thanks for the additional information. That is something that I've been trying to fix for some time. It mostly works for me and my X1 but I suspect the lower powered A1/P1 devices are still struggling with the solution.

JanUmbach, 2024-05-08:
I'm having the same problem with my P1S.

I would love to see a fix for that :) and btw nice plugin you've written so far! :D

jneilliii, 2024-05-08:
I need more details as to why this is happening. When you send a file to print from slicer OctoPrint should start to pick up that a print was started remotely, it then checks to see if the filename reported as printing from the printer is in OctoPrint's cached file list or not from initial connection. If it's not, it then requests an updated file list from the printer. When this happens for you guys are you seeing any messages in the terminal tab about printing <file name>? Enabling logging for the bambu items in OctoPrint's logging section and reproduce the issue and then share the octoprint.log and any other bambu logs generated.

JanUmbach, 2024-05-10:
When I start a print from slicer, octoprint will refresh the file list, but there is one problem. The file has 0 bytes, the console spams:

```
Send: N912 M105*45
Recv: T:209.00/ 170.00 B:64.00/ 65.00 @:64
Recv: ok
Send: N913 M27*29
Recv: Not SD printing
Recv: ok
Send: N914 M27*26
Recv: Not SD printing
Recv: ok
Recv: File opened: testttt.3mf  Size: 0
Recv: File selected
Recv: Done printing file
Send: N924 M105*40
Recv: T:200.00/ 170.00 B:64.00/ 65.00 @:64
Recv: ok
Send: N925 M27*24
Recv: Not SD printing
Recv: ok
Send: N926 M27*27
Recv: Not SD printing
Recv: ok
Recv: File opened: testttt.3mf  Size: 0
Recv: File selected
Recv: Done printing file
Send: N927 M105*43
Recv: T:197.00/ 170.00 B:64.00/ 65.00 @:64
Recv: ok
Send: N928 M27*21
Recv: Not SD printing
Recv: ok
Send: N929 M27*20
Recv: Not SD printing
Recv: ok
Recv: File opened: testttt.3mf  Size: 0
Recv: File selected
Recv: Done printing file
Send: N930 M20 L T*3
Recv: Begin file list
Recv: test.3mf 17848767 0x58a8b5ce "test.3mf"
Recv: testt.3mf 5588609 0x58aaa3b6 "testt.3mf"
Recv: testttt.3mf 5588609 0x58aaa419 "testttt.3mf"
Recv: End file list
Recv: ok
Send: N931 M105*44
Recv: T:196.00/ 170.00 B:64.00/ 65.00 @:64
Recv: ok
Send: N932 M27*30
Recv: Not SD printing
Recv: ok
Recv: File opened: testttt.3mf  Size: 5588609
Recv: File selected
Recv: SD printing byte 0/5588609
Send: N933 M27*31
Recv: SD printing byte 55886/5588609
Changing monitoring state from "Operational" to "Starting print from SD"
Recv: ok
Changing monitoring state from "Starting print from SD" to "Printing from SD"
Send: N934 M105*41
Recv: T:193.00/ 170.00 B:64.00/ 65.00 @:64
Recv: ok
Send: N935 M27*25
Recv: SD printing byte 55886/5588609
Recv: ok
Send: N936 M27*26
Recv: SD printing byte 55886/5588609
Recv: ok
Recv: SD printing byte 55886/5588609
[... 222 lines cut ...]
2024-05-10 13:24:19,693 - octoprint.plugins.bambu_printer.BambuPrinter - DEBUG - M105
2024-05-10 13:24:19,874 - octoprint.plugins.bambu_printer.BambuPrinter - DEBUG - M27
2024-05-10 13:24:20,875 - octoprint.plugins.bambu_printer.BambuPrinter - DEBUG - M27
2024-05-10 13:24:21,875 - octoprint.plugins.bambu_printer.BambuPrinter - DEBUG - M27
2024-05-10 13:24:22,875 - octoprint.plugins.bambu_printer.BambuPrinter - DEBUG - M27
2024-05-10 13:24:23,876 - octoprint.plugins.bambu_printer.BambuPrinter - DEBUG - M27
2024-05-10 13:24:24,693 - octoprint.plugins.bambu_printer.BambuPrinter - DEBUG - M105
2024-05-10 13:24:24,876 - octoprint.plugins.bambu_printer.BambuPrinter - DEBUG - M27
2024-05-10 13:24:25,876 - octoprint.plugins.bambu_printer.BambuPrinter - DEBUG - M27
2024-05-10 13:24:26,877 - octoprint.plugins.bambu_printer.BambuPrinter - DEBUG - M27
2024-05-10 13:24:27,878 - octoprint.plugins.bambu_printer.BambuPrinter - DEBUG - M27
2024-05-10 13:24:28,903 - octoprint.plugins.bambu_printer.BambuPrinter - DEBUG - M27
2024-05-10 13:24:29,694 - octoprint.plugins.bambu_printer.BambuPrinter - DEBUG - M105
2024-05-10 13:24:29,890 - octoprint.plugins.bambu_printer.BambuPrinter - DEBUG - M27
2024-05-10 13:24:30,891 - octoprint.plugins.bambu_printer.BambuPrinter - DEBUG - M27
2024-05-10 13:24:31,891 - octoprint.plugins.bambu_printer.BambuPrinter - DEBUG - M27
2024-05-10 13:24:32,891 - octoprint.plugins.bambu_printer.BambuPrinter - DEBUG - M27
```
Sorry, obico prints currently a lot of error messages

So maybe you're refreshing to early the print file?

Also my resend ratio is going crazy `0 / 5.4K (0%)`. Is it also a bug? I think the count gets faster up over time

Thank you for your time :)

jneilliii, 2024-05-11:
This is hard to figure out with a printer that works perfectly fine. I've added some additional logging info in a separate branch. Could you please install by copying/pasting the URL below into plugin manager > get more > ,,,from URL and clicking install, restart OctoPrint, reproduce the issue, and share the full log (you can use the paperclip button in comment editor)

```
https://github.com/jneilliii/OctoPrint-BambuPrinter/archive/refs/heads/bugfix/a1.zip
```

JanUmbach, 2024-05-12:
I have started two prints. `xyzCalibration_cube2` worked for whatever reason. And `xyzCalibration_TEST` is the same issue, so I have manually refreshed the files.

[octoprint.log](https://github.com/jneilliii/OctoPrint-BambuPrinter/files/15285471/octoprint.log)

jneilliii, 2024-05-12:
Can you share with me how that file that didn't work was created? I've heard of some weirdness with some of the test cubes in orca slicer being created oddly and some internal things within the file cause some naming inconsistencies. Can you upload the 3mf here (rename 3mf to zip before uploading).

JanUmbach, 2024-05-12:
[xyzCalibration_TEST.zip](https://github.com/jneilliii/OctoPrint-BambuPrinter/files/15286577/xyzCalibration_TEST.zip)

But this is not the original file as my printer is currently printing. But I can share it in a few hours :)

I also discovered a bug when I clicked on "Print Again" on the A1 display. Your plugin then looks for the file called "cached/myFile.3mf", but will never find it, as there are only files like "myFile.3mf". There's a 'cached' folder on the SD card, and that might lead to issues when reprinting. I'll also reproduce the issue and send you the logs in a few hours.

Thank you for your time :)

jneilliii, 2024-05-12:
OK, doesn't look like what I was thinking it was related to the cube weirdness of OrcaSlicer. What I don't understand is why it would work for the first one and not the second. The logs are basically showing the same process completed for each but on the second file, the file list returns a file size of 0. but then later down when you manually refresh the file list it shows the correct size. I'm actually really stumped here on this one. The only thing I can think of is a potential downloading files from cloud while the plugin is listing the files during the prepare stage of the print process. Since I don't use cloud mode on my X1 that may be what's happening here. I'd be curious if you were to switch to local only mode if the issue would persist or not.

JanUmbach, 2024-05-12:
Yes, I'm using the Cloud. Bu I can also try to switch to the local only mode. Maybe because the first one had a smaller file size and was faster processed than the second file with a bigger file size?

jneilliii, 2024-05-12:
One last question, are you using the plugin in local mode and the printer is in cloud mode, or do you have both connected to cloud?

JanUmbach, 2024-05-12:
The Plugin is connected with local mode, but the slicer and how I upload my prints are with cloud mode

jneilliii, 2024-05-12:
I think I have an idea to resolve this. I've just pushed an update to the bugfix/a1 branch to remove the "preparing" gcode_state from processing print start, and hopefully that will allow the printer time enough during that state for downloading the file before getting the file data over ftps. If you can try reinstalling with the same URL as before and restart OctoPrint and see if that gives more consistent results.

JanUmbach, 2024-05-12:
The fix worked just fine! I've tested two print files and every time the plugin started to recognize it with full file size. I will test more and also my P1S and will let you know for more testing.

Should I open a new issue for the "cached/" filename prefix problem or can I give my logs here in this issue?

jneilliii, 2024-05-12:
Good news then, hopefully that will help others as well, I'll go ahead and finalize a new release version with these changes incorporated. 

> Should I open a new issue for the "cached/" filename prefix problem or can I give my logs here in this issue?

Please open a separate issue for that with the logs of that print again issue. The plugin does see the files in cache folder, that's how it's able to get the file data from cloud uploaded files anyway, it just flattens all the files into a single root folder on SD card when listing them to OctoPrint.

closed 2024-05-12 by jneilliii