jneilliii/OctoPrint-BambuPrinter #11: Just wanted to thank you

aslabsalbeh, 2024-03-04:
I've been using the plugin for over 2 weeks and it's been amazing !

I found it by pure luck while searching github and I initially didn't know what it does. It allowed me to use Obico for monitoring all my prints on the A1. It's been such a relief to be able to use Octoprint again after parting ways with my Ender 3.

Do you mind sharing what you future plans are?

Do you think timelapse recording would be possible in the future or is this a Bambu limitation? I haven't figured it out yet.

Thank you and have a good one!

jneilliii, 2024-03-04:
Timelapse recording with the built-in timelapse functions work on a timed basis. I've been running it that way in my tests. For me and the X1C I have the ability to use go2rtc to restream the camera to OctoPrint, but on the A1 it's a little trickier. @synman has put together something for his A1 that should work for you as well, but I've heard framerate is really low here: https://github.com/synman/bambu-go2rtc.

As for plans for the plugin. I'm slowly developing it over my spare time and hoping to just get it to a stable state with print options similar to starting prints from the slicer, and AMS support.

synman, 2024-03-05:
what a nice issue lol :).  I wish everyone was as kind.

Yeah, my little go2rtc thing probably works well if you're looking to do further HA integration.  I know the HA scene loves go2rtc. 

If you're looking for something stand-alone, I'd actually go with my webcamd solution.   Configure this bad boy as a replacement for built-in octoprint webcam daemon and call it a day.

https://github.com/synman/webcamd/tree/bambu

closed 2024-03-13 by jneilliii