jneilliii/OctoPrint-BambuPrinter #13: Same file name showing when printing from Bambu Studio or Orca; file not uploading; multiple repeat alerts from Obico/Octoslack when starting

bdwilson, 2024-03-08:
No matter what file I upload, it seems to be always picking the last file that shows in the file listing.  It hasn't always done this, but seems to have started 2 days ago (that's when the last file name is showing in the Octoprint file list).  Even though the old name is used, the print still prints the correct item and has the correct time/length in Octoprint. 

Attached log shows finishing a print from Orca (name is not what is shown there).

Then shows me starting a new print from Bambu Studio (2024-03-08 16:26:01,969) which it appears to generate print resume -> starting print -> finishing -> operational a few times, then it prints the right file but still continues to show the old file from 2 days ago.  All prints show this file name.  All prints result in multiple notifications. 

[octoprint-debug.log](https://github.com/jneilliii/OctoPrint-BambuPrinter/files/14543441/octoprint-debug.log)

bdwilson, 2024-03-08:
They were already enabled. Here is the relevant snippet of the serial log. 
[plugin_bambu_printer_serial.log](https://github.com/jneilliii/OctoPrint-BambuPrinter/files/14543539/plugin_bambu_printer_serial.log)

jneilliii, 2024-03-08:
```
2024-03-08 16:21:02,662 - octoprint.plugins.bambu_printer.BambuPrinter - DEBUG - SD File Print: cache/latch v3 - gen3 - 4x pulls.3mf
```

perfect, after reviewing the octoprint log it seems you are cloud connected and not running in local only mode? I think the issue here is that the cached folder gets involved with cloud prints, so it's not tying back to the flattened file list that includes the cache sub-folder. I'll take a stab at a fix tonight after dinner to see if I can work that out.

bdwilson, 2024-03-08:
So it was installed before cloud connection came along. Use local is checked. I have not logged in via cloud and have all the local configs filled in.

jneilliii, 2024-03-08:
But your slicer is cloud connected otherwise files sent with the print button wouldn't go into cache sub folder.

bdwilson, 2024-03-08:
Ahh correct.

bdwilson, 2024-03-10:
Here is another debug log. 
[octoprint.log](https://github.com/jneilliii/OctoPrint-BambuPrinter/files/14548288/octoprint.log)