jneilliii/OctoPrint-BambuPrinter #14: Multiple print events are being triggered

aslabsalbeh, 2024-03-13:
Hi and thanks again for your great work.

I recently started having an issue when sending print commands to Bambu Studio. The printer appears to be receiving multiple commands at the same time?

I only figured this out because these are showing in Obico as ghost prints. I'm getting a number of zero minute prints, up to 6 times before the actual print is registered and recorded with the correct duration and timelapse. I'm using an A1.

Looking back at the print log from Feb last month, I don't have this issue.  If there's somewhere I can logs for you please let me know.

Thank you!

[image]

jneilliii, 2024-03-13:
Yeah, I've seen similar when starting a print where it flip flops a few times but haven't had the time to really dig into it. Are you starting the print from slicer, through handy app, or through OctoPrint? I have been informed recently that prints from slicer (if cloud connected) cause some strange issues with Obico.

aslabsalbeh, 2024-03-13:
I'm starting my prints from Bambu Slicer.

I've looked at Feb prints, & didn't see similar behavior.

Could be a recent update issue?

Cheers

jneilliii, 2024-03-13:
Possibly, or similar issue to #13

aslabsalbeh, 2024-03-19:
Hi again!

Not sure what happened since I last posted :)

It appears as though the plugin is no longer working! I reinstalled it from scratch, but it's not working as intended. 

When printing a model, it does show the printer temps, the name of the stl file etc... but I can't control the printer. It also doesn't show that it's actually printing, which is preventing Obico from working.

Printer currently printing, but this is what it's showing
[image]

aslabsalbeh, 2024-03-20:
I did additional testing.

Looks like the issue I described in my latest post is persistent.

I sliced a model and sent the print command through Bambu Slicer from my desktop.

Octoprint is connected to the printer, showing temp readings, and it shows the file name of the model I just printed. However, Octoprint doesn't say that the model is printing. It only updates the file name of the model (thus Obico doesn't detect failure because Octoprint doesn't show that I'm printing anything).

I cancelled the print, disconnected the printer from Octoprint, and reconnected. I started the print again from Bambu Studio and this time Octoprint detected that I'm printing. It not only shows the file name, but the print time and progress.

Didn't have this issue with the first few versions of the plugin.

Example on a "Cute Bunny" model that I was printing. Octorprint updated the file name when I clicked print in the slicer, but didn't reflect the start of the print.
[image]

jneilliii, 2024-03-20:
Please share octoprint and bambu logs (assuming you've enabled debug logging for all the bambu ones in OctoPrint's logging section. That will help debug. I need to know if you are cloud connected or not, and if the same thing happens if you send the print instead of using the print button and then refresh the file list in OctoPrint and start the print from there (making sure to start the one that is showing as being on the sd card).

aslabsalbeh, 2024-03-21:
I enabled the log for bambu plugin, so I'll provide you with that in the future.

I'm connected through the cloud. However the have enabled "Use Local Access, disable for cloud connection" because it was working fine before.

I started a new print like usual from Bambu Studio, and the 3mf file showed up in Octoprint when I clicked print from the slicer. But Octoprint didn't show that the printer is printing until I clicked on the SD Card icon under "Files" and clicked "Refresh SD files". Once I did that, Octoprint updated the status to Printing from SD and started showing the progress time.

jneilliii, 2024-03-21:
Yeah, so I suspect it's what I was thinking was the issue. The plugin/OctoPrint side doesn't know of the new file that was sent to bambu using the "print" button (this transfers via cloud and ends up in a cache folder on the printer side). I tried getting around that by re-retrieving the file list when that file isn't a "known file" in the plugin's cached list of files. I'm guessing if you look at the terminal tab when this is happening you'll see a bunch of SD printing 0/######.