jneilliii/OctoPrint-BambuPrinter #16: Unable to print in LAN Only mode to P1P - and potential Readme update.

Darthmonkey, 2024-03-15:
I'm not sure if printing to a P1P in Lan Only mode is possible as yet, if not, feel free to watch this post, and anyone who gets it working can update it. If it should be possible, the problems I'm having are detailed below, but basically the 'Recv: SD printing byte 0/29899' doesn't go above 0.

The following steps show in intricate detail, how to install, configure and connect to a P1P via a Pi4 on Octoprint. Feel free to use the information to append the 'Readme' if appropriate.
After the steps below, show issues in printing a file already on the SD card, in order to help with troubleshooting a basic vanilla installation of Pi4 | P1P | Octoprint | Lan Only Mode:
[image]
Happy to drop the text-only option too if required. Octoprint.log below:
[octoprint.log](https://github.com/jneilliii/OctoPrint-BambuPrinter/files/14610427/octoprint.log)

jneilliii, 2024-03-15:
could you please enable debug logging for all the bambu items in OctoPrint's logging section and then share the octoprint.log and bambu_serial.log after testing again. One thing I noticed, enable timelapse option is for the start print command, that would require that you have the camera on the P1P itself.

jneilliii, 2024-03-17:
please give the latest release 0.0.18 and report back if this process is working any better for you now.

Darthmonkey, 2024-03-18:
I installed the latest release, and it's now working!
I now have to look into how to choose an AMS slot, and get the camera working after my test print finishes.

jneilliii, 2024-03-18:
Great, currently AMS mapping is not possible and the sliced file has to have the proper assignments for use. That will change in the future after some tweaks by core OctoPrint team.