jneilliii/OctoPrint-BambuPrinter #18: P1S: PASV fails across subnets (with workaround)

disconn3ct, 2024-03-19:
Lots and lots of things don't work across subnets on the Bambu firmware, and one of them is `PASV`. It returns an IP of 0.0.0.0 and eventually the connection times out. (I can get debug logs if you want, but it hangs trying to connect to the provided port.)

Looking online, is it possible to adapt [this answer](https://stackoverflow.com/questions/56224932/python-ftplib-with-misconfigured-passive-connection)? It throws away the provided address and subs in the original:
```python
class FTP_TLS_IgnoreHost(ftplib.FTP_TLS):
    def makepasv(self):
        _, port = super().makepasv()
        return self.host, port

ftp = FTP_TLS_IgnoreHost('host', 'user', 'password')
```

jneilliii, 2024-03-19:
I've never had a network configuration to test this in, but I'm open to adapting the plugin to make it more compatible. If you are willing to test these changes locally and verify it works for you I can verify on my end it still works as expected in other areas.

jneilliii, 2024-03-19:
All the ftp stuff is in [here](https://github.com/jneilliii/OctoPrint-BambuPrinter/blob/master/octoprint_bambu_printer/ftpsclient/ftpsclient.py) and the connection is established in the plugin [here](https://github.com/jneilliii/OctoPrint-BambuPrinter/blob/908173214f5c053b89bb401abaf524b2efa1ff2c/octoprint_bambu_printer/__init__.py#L73). Ideally this would be configured as an additional option in settings instead of just forcing passive mode for all connections, as I'm not sure if that would cause any other additional issues.

disconn3ct, 2024-03-19:
Thanks! I'll dig into it a bit later. 
As far as configuration, I can't think of a situation where the Bambu would have a different address for one port (ftps) vs another (ftps-data) and not be incorrect in some way. The outgoing address should be the same for both even if it doesn't match what the printer thinks is local.
I'm hoping simply intercepting 0s is enough though.

disconn3ct, 2024-03-29:
I'm not sure what is going on with this. The ftp lib already handles `0.0.0.0` correctly, but curl can list files successfully and this can't. I'm going to have to put this aside for a bit though.

For anyone else who wants to pick it up, here is a quick cross-network curl that works. The logs hint that on connection refused curl falls back to using a single connection, but I haven't tried to convince this library to do that yet.

```
$ curl -vvvk ftps://10.128.11.146/ -u bblp:123456789
...
> EPSV
* Connect data stream passively
* ftp_perform ends with SECONDARY: 0
* TLSv1.2 (IN), TLS header, Supplemental data (23):
< 502 
* Failed EPSV attempt. Disabling EPSV
* TLSv1.2 (OUT), TLS header, Supplemental data (23):
> PASV
* TLSv1.2 (IN), TLS header, Supplemental data (23):
< 227 (0,0,0,0,7,232)
* Skip 0.0.0.0 for data connection, re-use 10.128.11.146 instead
*   Trying 10.128.11.146:2024...
* Connecting to 10.128.11.146 (10.128.11.146) port 2024
* Connected to 10.128.11.146 (10.128.11.146) port 990 (#0)
* TLSv1.2 (OUT), TLS header, Supplemental data (23):
> TYPE A
...
```