jneilliii/OctoPrint-BambuPrinter #19: 0.0.18 A1 disabled AI failure detection and disappeared from dashboard obico

hashoom8, 2024-04-06:
0.0.18 A1 disabled AI failure detection and disappeared from dashboard obico

something happened to my A1 after the update

obico no longer AI failure detection

jneilliii, 2024-04-06:
The plugin doesn't do anything related to snapshot or stream. Not sure what obico uses for it's progressing since I don't use it. Is it percentage based and/or the print start is not getting picked up? I know with the A1 that can happen if the slicer device tab initiated and starts reporting stuff, the printer basically stops sending info to the plugin. Usually disconnect and connect back will reregister the plugin to get that data.

hashoom8, 2024-04-06:
> The plugin doesn't do anything related to snapshot or stream. Not sure what obico uses for it's progressing since I don't use it. Is it percentage based and/or the print start is not getting picked [...]

by the way i send to your discord about my other huge problem