jneilliii/OctoPrint-BambuPrinter #20: how do i install this on my bambu printer I'm not very familiar with this and there is no clear install procedure.

CodingNoob27, 2024-04-06:


jneilliii, 2024-04-06:
This is not something you install on the printer. You install it inside an OctoPrint instance in plugin manager > get more button > ...fromURL, with the URL mentioned in the setup section.

https://github.com/jneilliii/OctoPrint-BambuPrinter#setup

closed 2024-04-06 by jneilliii

Darthmonkey, 2024-04-06:
I created some installation instructions which weren't used in a different post, I'm not sure why they were ignored, possibly due to being hidden behind a different question, or maybe they're incorrect, but if you're completely new to it, they may be of some help: [https://github.com/jneilliii/OctoPrint-BambuPrinter/issues/16](url)

jneilliii, 2024-04-06:
@Darthmonkey because it was just a screenshot of your instructions it didn't lend itself to an easy import into the readme, or linked to from the readme. If you could provide the raw notes as a comment I could easily add it then.

Darthmonkey, 2024-04-07:
Standard OctoPi installation for a Pi4 (other Pi installs will be similar)

  1.  Use Raspberry Pi Imager to install OctoPi (stable) from 'Other Specific Purpose OS' | '3D Printing'.
  2.  Connect a suitably sized SD card to the PC, and select the 'Storage' dropdown to find it.
  3.  Edit the connection settings if required to add Wireless information, and username  and passwords.
  4.  Add the SD card to a Pi, and boot it up with a  suitable power supply (this is important as issues may arise if insufficient power is provided during printing).
  5.  Allow a few minutes (at least 2.5 for a quick Pi4 and decent SD card) for the Pi to boot and connect to your wireless network.
  6.  Find the IP on your router or by scanning using something like Angry IP scanner for 'OctoPi'
  7.  Connect a USB webcam now if you have one to one of the older style white USB ports.
  8.  Browse to the IP address, and go through the setup.
     *   The test should find the webcam connected to the Pi, but don't worry if it's not found at this point.
     *   A 'Default Printer Profile' is not required at this stage, simply use the 'Generic RepRap Printer' defaults.
     *   Read all the other prompts, and Finish
  9.  Run any prompted updates, and revise your power supply if it's complaining about low power. PC or standard USB power supply is usually not enough.

Configure 'OctoPrint-BambuPrinter'

  1.  Click the spanner icon, and select 'Plugin Manager' | '+ Get More' | Paste the URL 'https://github.com/jneilliii/OctoPrint-BambuPrinter/archive/master.zip'
     *   Check for any issues in the text box while installing, this will be recorded in a log but copy any red text here to investigate further. | Close | Close | Save | Restart Now | Proceed.
  2.  Reload to log back in, and click the spanner icon again, then 'Bambu Printer'
     *   Choose your Device Type from the dropdown list - Note: Only one Bambu printer can be connected in version 0.0.17
     *   Locate your Bambu Printer IP on your LAN by searching for the name or finding it on the printer.
        *   Ideally reserve the IP on your router so it keeps it. Add this in the IP Address section.
     *   Add the printer Serial Number and Access codes which can be found on the printer.
  3.  If using LAN Only Mode on the printer, leave the 'Use Local Access…' box checked.
  4.  Check the Use AMS box if you have one.
  5.  Leave 'Enable timelapse' until you have the webcam working correctly.
  6.  Save.

Connecting to a Bambu printer

The 'Connection' box will stop showing red, and allow you to connect, select 'Save connection settings' and 'Auto-connect on server startup', the 'Connect'.

  *   It should now connect to the printer and list all the 3mf and gcode already on the printer SD Card.
  *   If using Home Assistant, existing connections will stop updating while it's connected (with the possible exception of the camera).

--- End of proposed 'Installation - Configuration - Connection' steps. ---

Troubleshooting steps

To test basic operation, sort by size, and print a small print already on the SD card.

  1.  Spanner icon | Logging | Logging Levels
     *   Select a logger | 'octoprint' | Level - Debug
     *   Save
  2.  Note in the 'Terminal' tab that nothing is being printed to the printer

Recv: ok

Send: N8 M23 /pa_lin~1.3mf*105

Recv: File opened: pa_line_calib_mode.gcode.3mf  Size: 29899

Recv: File selected

Changing monitoring state from "Operational" to "Starting print from SD"

Recv: ok

Send: N9 M23 /pa_lin~1.3mf*104

Recv: File opened: pa_line_calib_mode.gcode.3mf  Size: 29899

Recv: File selected

Recv: ok

Send: N10 M24*36

Recv: ok

Changing monitoring state from "Starting print from SD" to "Printing from SD"

[...]

Recv: ok

Send: N12 M27*37

Recv: SD printing byte 0/29899

Recv: ok

Send: N13 M27*36

Recv: SD printing byte 0/29899

Recv: ok

  1.  Note in the 'State' box:

Error stage:

State: Printing from SD

Resend ratio: 0 / 108 (0%)

  1.  Download the 'octoprint.log' from Spanner icon | Logging | check 'octoprint.log' | Download icon | (keep if prompted)
     *   View the log file in a viewer such as CMTrace which highlights error messages, and note that nothing of interest other than

Darthmonkey, 2024-04-07:
**Standard OctoPi installation for a Pi4 (other Pi installs will be similar)**

  1.  Use Raspberry Pi Imager to install OctoPi (stable) from 'Other Specific Purpose OS' | '3D Printing'.
  2.  Connect a suitably sized SD card to the PC, and select the 'Storage' dropdown to find it.
  3.  Edit the connection settings if required to add Wireless information, and username  and passwords.
  4.  Add the SD card to a Pi, and boot it up with a  suitable power supply (this is important as issues may arise if insufficient power is provided during printing).
  5.  Allow a few minutes (at least 2.5 for a quick Pi4 and decent SD card) for the Pi to boot and connect to your wireless network.
  6.  Find the IP on your router or by scanning using something like Angry IP scanner for 'OctoPi'
  7.  Connect a USB webcam now if you have one to one of the older style white USB ports.
  8.  Browse to the IP address, and go through the setup.
     *   The test should find the webcam connected to the Pi, but don't worry if it's not found at this point.
     *   A 'Default Printer Profile' is not required at this stage, simply use the 'Generic RepRap Printer' defaults.
     *   Read all the other prompts, and Finish
  9.  Run any prompted updates, and revise your power supply if it's complaining about low power. PC or standard USB power supply is usually not enough.

**Configure 'OctoPrint-BambuPrinter'**

  1.  Click the spanner icon, and select 'Plugin Manager' | '+ Get More' | Paste the URL 'https://github.com/jneilliii/OctoPrint-BambuPrinter/archive/master.zip'
     *   Check for any issues in the text box while installing, this will be recorded in a log but copy any red text here to investigate further. | Close | Close | Save | Restart Now | Proceed.
  2.  Reload to log back in, and click the spanner icon again, then 'Bambu Printer'
     *   Choose your Device Type from the dropdown list - Note: Only one Bambu printer can be connected in version 0.0.17
     *   Locate your Bambu Printer IP on your LAN by searching for the name or finding it on the printer.
        *   Ideally reserve the IP on your router so it keeps it. Add this in the IP Address section.
     *   Add the printer Serial Number and Access codes which can be found on the printer.
  3.  If using LAN Only Mode on the printer, leave the 'Use Local Access…' box checked.
  4.  Check the Use AMS box if you have one.
  5.  Leave 'Enable timelapse' until you have the webcam working correctly.
  6.  Save.

**Connecting to a Bambu printer**

The 'Connection' box will stop showing red, and allow you to connect, select 'Save connection settings' and 'Auto-connect on server start up', the 'Connect'.

  *   It should now connect to the printer and list all the 3mf and gcode already on the printer SD Card.
  *   If using Home Assistant, existing connections will stop updating while it's connected (with the possible exception of the camera).

Darthmonkey, 2024-04-07:
> @Darthmonkey because it was just a screenshot of your instructions it didn't lend itself to an easy import into the readme, or linked to from the readme. If you could provide the raw notes as a comment I could easily add it then.

I did mention that if required I could post the raw text. Not sure why OneNote doesn't copy text by default, I suppose I was being lazy. ;)