jneilliii/OctoPrint-BambuPrinter #22: auto connect not working

hashoom8, 2024-04-27:
if the printer was powered off , and than turned on  , octoprint doesn't connect automatically
i must click on connect every time powering on the printer 

my old ender 5 it connect automatically

jneilliii, 2024-04-27:
true, there's no reattempt at connecting once a connection is lost. with a physical connection to the pi on your ender 5 that can be detected easily by the OS. For this plugin it would require polling the ip address after a connection is lost to know when the printer comes back on. I personally never power my printer off so not sure when I would get around to adding this feature when you can just press the connect button.

hashoom8, 2024-04-27:
> true, there's no reattempt at connecting once a connection is lost. with a physical connection to the pi on your ender 5 that can be detected easily by the OS. For this plugin it would require [...]

not everyone using this plug in the same way everyone using it differently for me I am using a cloud connection and what I do is using AI failure detection so sometimes i start a print on the slicer and the print failed and become a huge miss because I forgot to press the connect button

hashoom8, 2024-04-27:
or you can add an option in the settings to check the IP address every 5 minutes and if it's available reconnect automatically

jneilliii, 2024-04-27:
> or you can add an option in the settings to check the IP address every 5 minutes and if it's available reconnect automatically

yeah, this was kind of how I would think to do it. but instead of making it an option would just automatically start polling the IP on disconnect say every minute. 

BTW, which device do you have?

hashoom8, 2024-04-27:
> yeah, this was kind of how I would think to do it. but instead of making it an option would just automatically start polling the IP on disconnect say every minute.
>
> BTW, which device do you have?

raspberry pi 4 8gb and bambu lab A1 + pi cam V3

hashoom8, 2024-04-27:
> yeah, this was kind of how I would think to do it. but instead of making it an option would just automatically start polling the IP on disconnect say every minute.
>
> BTW, which device do you have?

another problem after octoprint update to 1.10.1 ,  can't connect to the printer

jneilliii, 2024-04-27:
There is no OctoPrint 1.10.1 and I've been using 1.10.0 for quite some time without issue. The A1 is very finicky about who is connected and who isn't. If you open the device tab of slicer and start getting data from the printer the printer will stop sending info to this plugin until a reconnect. That's unfortunately an issue with the printer itself and the same reconnect watchdog we were discussing is what would hopefully fix that too.

hashoom8, 2024-04-27:
> There is no OctoPrint 1.10.1 and I've been using 1.10.0 for quite some time without issue. The A1 is very finicky about who is connected and who isn't. If you open the device tab of slicer and start [...]

sorry 1.10.0 
[image]

jneilliii, 2024-04-27:
so it's dying on attempting to read the file list from the printer via ftps.

hashoom8, 2024-04-28:
> so it's dying on attempting to read the file list from the printer via ftps.

yes , that happens after the update 1.10.0

jneilliii, 2024-04-28:
Can you try enabling this checkbox in OctoPrint's settings, restarting OctoPrint and let me know if it makes any difference? 

[image]

hashoom8, 2024-04-28:
> Can you try enabling this checkbox in OctoPrint's settings, restarting OctoPrint and let me know if it makes any difference? [...]

it's checked by default
[image]

jneilliii, 2024-04-28:
Ah, wasn't sure if I had unchecked it or not. In my current setup it was unchecked actually.

hashoom8, 2024-04-28:
> Ah, wasn't sure if I had unchecked it or not. In my current setup it was unchecked actually.

i did try to uncheck it , and still no connect

jneilliii, 2024-04-28:
enable debug logging in OctoPrint's logging section for all plugin.bambu_printer options and lets see if there are any hints in octoprint.log or the bambu logs that get generated.

hashoom8, 2024-04-28:
> enable debug logging in OctoPrint's logging section for all plugin.bambu_printer options and lets see if there are any hints in octoprint.log or the bambu logs that get generated.

[serial.log](https://github.com/jneilliii/OctoPrint-BambuPrinter/files/15140823/serial.log)
[plugin_bambu_printer_serial.log](https://github.com/jneilliii/OctoPrint-BambuPrinter/files/15140824/plugin_bambu_printer_serial.log)

hashoom8, 2024-04-28:
> enable debug logging in OctoPrint's logging section for all plugin.bambu_printer options and lets see if there are any hints in octoprint.log or the bambu logs that get generated.

https://drive.google.com/file/d/1gcZBUWKtI16zXkfxHTNNZGUvMDmoN7aJ/view?usp=sharing

hashoom8, 2024-04-28:
> enable debug logging in OctoPrint's logging section for all plugin.bambu_printer options and lets see if there are any hints in octoprint.log or the bambu logs that get generated.

[octoprint (1).log](https://github.com/jneilliii/OctoPrint-BambuPrinter/files/15142234/octoprint.1.log)

jneilliii, 2024-04-28:
You're using cloud connection to bambu online? 

> 2024-04-28 16:05:36,272 - octoprint.plugins.bambu_printer.BambuPrinter - DEBUG - connecting via local mqtt: False

and looking at some of the following errors, seems you are experiencing a network related issue with the underlying python module I'm using. 

> 2024-04-28 16:05:39,457 - pybambu - ERROR - A1: Exception. Type: <class 'OSError'> Args: [Errno 113] No route to host

hashoom8, 2024-04-28:
> You're using cloud connection to bambu online?
>
>
> and looking at some of the following errors, seems you are experiencing a network related issue with the underlying python module I'm using.

yes cloud connection  , so how can i fix it

hashoom8, 2024-04-29:
> You're using cloud connection to bambu online?
>
>
> and looking at some of the following errors, seems you are experiencing a network related issue with the underlying python module I'm using.

problem solved , i restart all my home network