jneilliii/OctoPrint-BambuPrinter #23: can you make octoprint read gcode.3mf file

hashoom8, 2024-04-28:
can you make octoprint read gcode.3mf file

so I’m trying to take time-lapse with octoprint , but octoprint can’t read gcode.3mf file

also I’m thinking to add trigger switch on the Purge Wiper and connect it to raspberry pi gipo pins

i print multi color AMS through bambu slicer because not working for me if i print through octoprint

jneilliii, 2024-04-28:
The plugin supports AMS but it has to be sliced with the correct color/filament mapping. It also should upload 3mf file to your printer, but you have to upload to SD card in OctoPrint but with the file listing/ftps connection problems you're experiencing in the other issue will also effect your ability to upload.

jneilliii, 2024-04-28:
Also, plugin will only print files from SD card in the file list. Timelapse only works on timed option, not on z change option.

hashoom8, 2024-04-29:
> Also, plugin will only print files from SD card in the file list. Timelapse only works on timed option, not on z change option.

How does octoprint know the progress of the print , because what I am thinking is how to send a gcode from the printer to octoprint terminal

and that gcode can be anything , for example M73 ; snap-shot

I know how to add the gcode in the slicer , and i know how to configure timelapse with gcode

jneilliii, 2024-04-29:
Well, if the connection between the plugin and printer is working then the printer tells it print progress and the plugin adapts that for SD printing progress by using the filesize.

hashoom8, 2024-04-29:
> Well, if the connection between the plugin and printer is working then the printer tells it print progress and the plugin adapts that for SD printing progress by using the filesize.

is there is any chance to make octoprint read the gcode while it's printing

jneilliii, 2024-04-29:
There is not because of the way the Bambu printer functions.

ulmeck, 2024-05-15:
Would there be a way to pull the currently printing gcode via ftps, and then force feed that to octoprint as the current gcode? That would give something like print time genius and the gcode viewer to chew on. Granted, that's probably a lot further into the guts of octoprint than one would want to go to for this. 

Essentially it would be 'fixing' the problem of SD card printing being completely detached from the gcode analysis in octoprint. But that might be a heavy lift.

jneilliii, 2024-05-15:
Unfortunately things like current position aren't reported by the firmware, so I fear that's not going to change. Although it's technically possible to extract the gcode file from the 3mf for processing locally in OctoPrint, there's not a good approach available for handling command buffering to the printer, and even acceptance of commands sent to the printer. Bambu doesn't have the same kind of direct serial communication that you would have with Marlin, Klipper, Mainsail, etc. firmware.

ulmeck, 2024-05-15:
Oh, no. Nothing like that. I was just thinking about the peripheral stuff that wants to see the gcode on the octoprint side. I know that the bambu doesn't have a serial interface or anything that lets you poke the gcode into it.  

Specifically, I'm thinking of things like print genius (for print prediction), preheat button (which gets the initial temps from the gcode file), and the gcode preview tab.

jneilliii, 2024-05-15:
The problem comes down to a disconnect between the SD card files and the locally stored files. There's no good way within the OctoPrint framework to share data between the two storage objects (that I know of). I have been considering ways to do this though, like extracting thumbnails to show in the file list like my slicer thumbnails plugin does. The only missing piece would be expanding the metadata of SD card files and having that displayed, which currently in OctoPrint is solely based on the information presented with M20 L T command, which just gives file name, file size and timestamp.

ulmeck, 2024-05-15:
Huh. I did not realize you aren't using the sftp interface on the bambu. That would get access, but I have no idea ho hard it would be to implement.

jneilliii, 2024-05-16:
I am using ftps to send the file to the printer when you upload via OctoPrint and retrieve file listings with that information that shows.