jneilliii/OctoPrint-BambuPrinter #25: The current print isn't being recognized by this plugin when I try to use the 'print again' function

JanUmbach, 2024-05-12:
My environment:
Printer: A1
OctoPrint-BambuPrinter Plugin is running localy, but Slicer will send data over the cloud.

When I start a print from my Bambu Studio Slicer, everything works fine.

When I start a print from the display, it also works fine. But after a print is finished, I have the option to click "Print Again" at the A1 display, and now he adds a 'cache/' prefix to the filename instruction. The plugin is searching for the filename e.g. 'cache/myFile.3mf', but it won't find it.

Terminal says: `Recv: cache/testt.3mf open failed`
[octoprint.log](https://github.com/jneilliii/OctoPrint-BambuPrinter/files/15287230/octoprint.log)

jneilliii, 2024-05-12:
I think I may have fixed the issue in the above commit. Could you please test on your end by installing the plugin using the URL below in Plugin Manager > Get More > ...from URL and report back?

```
https://github.com/jneilliii/OctoPrint-BambuPrinter/archive/refs/heads/bugfix/print_again_cache.zip
```

JanUmbach, 2024-05-14:
Yes that fixes that issue, thanks :)

PR #27 (2024-05-14): 0.0.20
* fixes issue related to printing a cloud print again and subtask_name including full path to file, closes #25
closed 2024-05-14 by jneilliii