jneilliii/OctoPrint-BambuPrinter #26: Unable to connect to P1P

navytrav, 2024-05-13:
I dug out and set up an older raspi to give this a try, but it will not connect. I may be missing something, and any help getting it working would be appreciated.

Fresh install of OctoPi on Raspberry Zero 2 W.
P1P in cloud mode.
No errors in plugin console log.
Clicking login on the configuration page does not visibly do anything except remove the entered auth token.
"pybambu - WARNING - On Disconnect: Disconnected from Broker: 5" showing in Octoprint log every few seconds.
"pybambu - ERROR - P1S: Socket error: [Errno 104] Connection reset by peer" logged on connection attempt.
No logs are created after enabling info level logging on all octoprint.plugins.bambu_printer options.

jneilliii, 2024-05-13:
Logging needs to be set to debug to get anything in octoprint.log from the plugin. I would recommend trying not in cloud mode to begin with. The printer should still allow for connecting locally outside of the cloud connection and just needs the IP address of your printer and the access code. I do not know where in the printer menu that information exists since I don't have a P1P, but it should be in there somewhere.

navytrav, 2024-05-13:
I think I may have figured it out. I was able to connect, but it disconnected fairly quickly. Its mid long print, so I can't fully test.

Basic problem is I didn't double check the access code, just thought the picture I had from before was still accurate.

jneilliii, 2024-05-14:
Just checking in to see if you are still having connection issues or not before closing?

navytrav, 2024-05-14:
Issue caused by my own mistake

closed 2024-05-14 by navytrav