jneilliii/OctoPrint-BambuPrinter #28: Feature Request - Expose cache and timelapse files via octoprint

ulmeck, 2024-05-15:
Right now, it looks like the file interface only shows files n the root on the Bambu sd card. Could it be possible to 

a) expose the 'cache' directory, which has old jobs in it 

and 

b) provide access to the camera timelapse data in the 'timelapse' folder. I don't know if there's a way to integrate it into the timelapse tab, but it would be a very nice thing to have.

jneilliii, 2024-05-15:
a. The files in cached folder are actually listed, but the entire file list is flattened in OctoPrint's UI. 

b. That is an interesting idea that I never thought of since I don't use the timelapse feature of the printer. I can definitely look into the possibility of injecting the information into the timelapse tab of OctoPrint.

jneilliii, 2024-05-18:
New version 0.0.21 has been released with initial timelapse support. You'll see the files listed on the Timelapse tab of OctoPrint at the bottom.

closed 2024-05-18 by jneilliii