jneilliii/OctoPrint-BambuPrinter #30: Timelapse thumbnails broken on P1P/P1S

ulmeck, 2024-05-25:
Looks like the thumbnail name is dependent on the timelapse file being an MP4. 

from __init.py__

` "thumbnail": "/plugin/bambu_printer/thumbnail/" + filename.replace(".mp4", ".jpg"),`

ulmeck, 2024-05-25:
It also looks like the download portion is broken somehow.

`2024-05-25 01:45:51 - tornado.access - INFO - 200 GET /sockjs/info?t=1716601542159 (192.168.100.150) 4.28ms
2024-05-25 01:45:51 - tornado.access - INFO - 302 GET /plugin/bambu_printer/timelapse/video_2024-05-24_17-27-38.avi (192.168.100.150) 34.05ms
2024-05-25 01:45:51 - tornado.access - INFO - 200 GET /plugin/bambu_printer/download/timelapse/video_2024-05-24_17-27-38.avi (192.168.100.150) 2.64ms
2024-05-25 01:45:54 - tornado.access - INFO - 101 GET /sockjs/785/zoa2aj3z/websocket (192.168.100.150) 2.10ms
2024-05-25 01:45:54 - tornado.access - INFO - 200 POST /api/login (192.168.100.150) 37.94ms
2024-05-25 01:45:54 - tornado.access - INFO - 200 GET /api/files?recursive=true (192.168.100.150) 35.20ms
`

All that gets saved is a 0 byte file.

jneilliii, 2024-05-25:
ah good point, forgot about that bit before. I'll get that fixed.

closed 2024-05-25 by jneilliii