jneilliii/OctoPrint-BambuPrinter #32: A1 Mini Webcam

TLMcNulty, 2024-06-19:
I've done some searching but haven't found anything. The Bambu plugin and workflow works well once you understand how to use this plugin, but I'm familiar with the RTC streams in the new camera stack stuff in octoprint, and was wondering if there was any documentation on getting the A1 Mini's webcam into a platform like octoprint. May be out of scope for this plugin but I was hoping it'd be possible to add to my multicam setup.

TLMcNulty, 2024-06-19:
A few resources I've come across:

https://github.com/synman/bambu-go2rtc
https://github.com/bambulab/BambuStudio/issues/1536#issuecomment-1811916472
https://github.com/slynn1324/BambuP1Streamer
https://github.com/synman/webcamd/tree/bambu

jneilliii, 2024-06-19:
Yeah, synman's approach is something we kind of worked out together and probably the only way to integrate well with OctoPrint. I actually use go2rtc and point to my x1c's rtsp stream, then use the HLS link for that stream in OctoPrint's classic webcam settings.

TLMcNulty, 2024-06-19:
Very impressive. I was able to get the stream into octoprint as I described wanting. Running the go2rtc binary gets video and opens the ports locally as described. We just need to formalize some sort of approach for defining a Bambu printer in Octoprint and setting up go2rtc.

jneilliii, 2024-06-20:
I had it running as a service using these steps: https://gist.github.com/jneilliii/93b412cb0bbf6b7bfd76f7e10d612f24

you would just have to deal with any path differences where you saved the go2rtc binaries/config. it assumes pi username in `ExecStart`

secco04, 2024-10-26:
@MrDrache333  hey hi thanks for that but still i have trouble to get an image out .... 
i assume the "hostname" must be the printers ip address?

Zeunas, 2024-11-18:
> Yeah, synman's approach is something we kind of worked out together and probably the only way to integrate well with OctoPrint. I actually use go2rtc and point to my x1c's rtsp stream, then use the HLS link for that stream in OctoPrint's classic webcam settings.

Hey, any chance you could explain how to install this? https://github.com/synman/webcamd/tree/bambu
Thanks!

kalinjul, 2024-12-25:
@MrDrache333 Your docker compose file works perfectly fine for me, thanks!
Using Printer's ip address as HOSTNAME and accesscode as PASSWORD.