jneilliii/OctoPrint-BambuPrinter #35: possible A1 issues

jneilliii, 2024-07-30:
cross posting issue from community forum, not sure if it's related to the recent refactor by @Valemos yet or not. 

https://community.octoprint.org/t/octoprint-for-bambu-lab-a1-support-available/56294/84?u=jneilliii

MisterTee8227, 2024-07-30:
[octoprint-systeminfo-20240731004111.zip](https://github.com/user-attachments/files/16434362/octoprint-systeminfo-20240731004111.zip)

MisterTee8227, 2024-07-30:
Wait it appears to be working now.  After I clocked the System info screen, the temperature and print status was working.  Once this print finish I'll try sending something from Orcaslicer to it.

MisterTee8227, 2024-07-31:
Just to update you, while the print status and temperature appear to be working now, I can't send a job to it.  The status is showing but I am unable to print from Orca or Bambu Studio.  Its possible I have that configured incorrectly but when I hit print, my P1S does fire up for a second then gives and error.

```
`Serial` timeout while writing to serial port, trying again.
Unexpected error while writing to serial port: SerialTimeoutException: '' @ comm.py:_do_send_without_checksum:5057
Please see https://faq.octoprint.org/serialerror for possible reasons of this.
Changing monitoring state from "Printing" to "Offline after error"
Closing down send loop
Connection closed, closing down monitor
```

jneilliii, 2024-07-31:
Please upload a system info bundle, you can use the paper clip to attach the complete zip file.

MisterTee8227, 2024-07-31:
[octoprint-systeminfo-20240731022451.zip](https://github.com/user-attachments/files/16435040/octoprint-systeminfo-20240731022451.zip)
 BTW I noticed when going back into Octoprint to get this for you that its not showing my print status again even though I am printing.  Nothing has changed that I know of.

jneilliii, 2024-07-31:
Are you trying to print locally uploaded files from OctoPrint on your printer? The way the plugin works with Bambu is by mimicking a virtual sd card, so if you're not trying to start a print that is on the SD card the transferred data/plugin may do weird stuff. You can still start files from the OctoPrint interface, but they need to be SD card files as indicated by this icon. 

[image]

you can also filter the file list to only show these files. 

[image]

Also on Serial Connection settings > firmware & protocol tab you may want to add M20 to the long running commands list in the advanced options section.

[image]

It would also be helpful to know if you are cloud connected or not for both the printer and the plugin's settings. I know you mentioned xtouch and was afraid to upgrade firmware, so I am making the assumption that you are running in local only mode.

jneilliii, 2024-07-31:
The error that you were getting with the M220 command was resolved in the above commit and has been released as version 0.1.1.

MisterTee8227, 2024-07-31:
I am trying to print directly from Bambu studio.  In my testing I was using Orca slicer.  After hitting print, the printer does wake up and start moving.  It stops shortly there after and errors out.  I don't do anything with the SD card.  Every print comes from Bambu studio or Orca.  I am Cloud connected and would prefer to stay that way since I like the ability to skip objects using Bambu Handy if there is a problem with one of the printed models.  As for the firmware update, I did finally load the update last night thinking that fix my issue.  It didn't, but it also didn't break my Xtouch either so that's good.  My Xtouch goes through the cloud as well.  Nothing is running locally.

Are you saying you fixed the issue and that I need to download a new version?  Also what should I be typing into the region box when configuring the plugin.  I have tried USA, usa, Usa, China, CHINA, and china.  None seem to make a difference.  I get a key when I click login no matter what I put in there so maybe it doesn't matter?  Someone in the comments said that they got it working by changing the region to China so I wanted to try it.

jneilliii, 2024-07-31:
Ok, P1/A1 devices do not like mixing local and cloud modes. If you are using cloud mode for the printer you need to use cloud mode for the plugin. Your region is based on where your Bambu login is associated. If you don't live in China, that's not what you want to use. 

I only fixed the error that was showing in your log for attempting to change the feedrate. That could have been causing a disconnect state, so if you could upgrade to the latest version and try to connect to cloud mode in the plugin's settings and report back that would be helpful.

MisterTee8227, 2024-07-31:
I updated to version 0.1.1 and the temperatures are showing correctly, although there is a few minute delay before you see it.  I attempted to send a print from Bambu Studio to Octoprint and initially it looked like it would work.  The printer started heating and tried to print.  First I got this error: T1000 something about the tool doesn't exist.  I went to lunch thinking it would print or fail but apparently the head moved around so much that the tool cover fell off and nothing was printed.  Something is definately not right but i'm not sure if its my Orcaslicer config or a problem with Octoprint or the plugin for it.
``

MisterTee8227, 2024-07-31:
> Ok, P1/A1 devices do not like mixing local and cloud modes. If you are using cloud mode for the printer you need to use cloud mode for the plugin. Your region is based on where your Bambu login is [...]

I believe I'm using cloud mode for everything.  printer is cloud, xtouch is cloud, bambu studio is cloud and the Octoprint plugin is cloud.  Thats why I asked about the region.  Is something in my logs showing that I am doing something locally?

jneilliii, 2024-07-31:
> I attempted to send a print from Bambu Studio to Octoprint

This should be sending directly to the printer, not to OctoPrint. I don't even know how you can configure Orca/Bambu Slicers to send to OctoPrint directly, so I'm confused on this statement. 

One thing you should double-check, make sure in OctoPrint you don't have any gcode scripts configured in settings as well.

[image]

jneilliii, 2024-07-31:
oh, one other thing I just thought of since you are cloud connected. Make sure the option is enabled to cache files on the printer. not sure where that setting is on the P1S but know that does impact the ability for the plugin to see the file listings.

MisterTee8227, 2024-07-31:
> oh, one other thing I just thought of since you are cloud connected. Make sure the option is enabled to cache files on the printer. not sure where that setting is on the P1S but know that does impact the ability for the plugin to see the file listings.

The file listing is the one thing that has worked from the beginning so I'm not going to mess with that at go and break it.  I may look for the setting just to see where it is.

MisterTee8227, 2024-07-31:
> > I attempted to send a print from Bambu Studio to Octoprint
> 
> This should be sending directly to the printer, not to OctoPrint. I don't even know how you can configure Orca/Bambu Slicers to send to OctoPrint directly, so I'm confused on this statement.

I saw a video on youtube and from that I kind of guessed on some stuff.  Based on what it did today, I would say I guessed wrong.  Printing directly to the printer makes much more sense.  I used to have an Ender 5 and the only way to print without usb was to go to Octoprint.  Old habits die hard and this makes way more sense.   I spent a good deal of time last night trying to figure out how to print to Octoprint.  What a waste of time.

MisterTee8227, 2024-07-31:
I'm going to give it a good test now that I know the proper printing method.  I'll let you know what works and what doesn't.  Thanks so much for your help.

jneilliii, 2024-07-31:
> Old habits die hard and this makes way more sense. I spent a good deal of time last night trying to figure out how to print to Octoprint. What a waste of time.

Oh yeah, I guess it's not really clear or well defined how that works. It's definitely a different methodology to the classic OctoPrint setup.

MisterTee8227, 2024-07-31:
speaking of old habits.  I wiped out my old Octoprint install from years ago and in it I had a script that would disable the autofocus.  I think I had it run from Rc.d but its really foggy.  I believe the focus level was defined in a conf file.  I can't find any backup of it and I have the same problem.  I wish I could remember what I did to fix it last time.  Has the print head moves the camera (which is a lifecam HD-5001) will constantly try to auto focus and the result is it never really focuses and its just blurry.  I don't even remember where that conf file would have been.  Do you happen to know where I should look to find this info again?

I have been testing and so far so good.  I paused the print and resumed and it worked as it should.  I haven't tried moving any of the axis yet but the fan  off/on buttons work.  The recent file display shows up perfectly.  The gcode viewer is blank.  Is that something that is supposed to work?

jneilliii, 2024-08-01:
GCode Viewer won't be able to work because there is no direct access to the file since it's stored on the printer and presented to OctoPrint via virtual SD card. It's the same thing that happens with gcode files on SD card for other printers. 

relative your focus question, you might want to check out the camera settings plugin. It will allow you to modify stuff from OctoPrint's settings and store a profile to auto-load with OctoPrint startup.

closed 2024-08-01 by jneilliii

MisterTee8227, 2024-08-01:
I found the camera settings control panel at http://octopi.local/webcam/control.  This gave me what I want but the setting revert after every reboot.  It says to make it persist to set a persist option.  I believe this is set in /boot/camera-streamer/usb-default.conf but I can't seem to save the changes using vi or upload a changed file with File Zilla over SSH.  I believe this is the answer but I'm hung up on permissions for the file.  If I am mistaken in anything I said, please let me know.

jneilliii, 2024-08-01:
try `sudo nano /boot/camera-streamer/usb-default.conf` and make your changes. ctrl + s to save, ctrl + x to exit. and then `sudo service camera-streamer restart`.

One other thing I just thought of about sending direct to OctoPrint. You can actually do that, but you just have to make sure it's sending to the SD card. There's no real reason to do that though because the plugin will then just upload the file to the printer, so sending direct from slicer removes an extra step.

MisterTee8227, 2024-08-01:
> try `sudo nano /boot/camera-streamer/usb-default.conf` and make your changes. ctrl + s to save, ctrl + x to exit. and then `sudo service camera-streamer restart`.

Here are the changes I made.  After the restart the webcam stopped working.  I did a full reboot and it still gives me the error: "Webcam stream not loaded"

`
# Additional options.
OPTIONS= --camera-options persist
`

MisterTee8227, 2024-08-01:
I figured it out.  I misunderstood.  You have to list each option out.  got it working now.  thanks for your help!