jneilliii/OctoPrint-BambuPrinter #36: A1 Mini pausing print does nothing octo print reports pausing status

Kitki30, 2024-08-02:
I tried to use pinozcam plugin to prevent failed prints but when it started to pause the print it never paaused but octo reported pausing status

Kitki30, 2024-08-02:
i found out that you need to set pausing command to
M400 U1

Kitki30, 2024-08-02:
but fans are 100%

jneilliii, 2024-08-02:
You should be able to add the commands to turn off fans on pause in OctoPrint's gcode scripts section. Probably want to make sure to turn them back on in resume.

jneilliii, 2024-08-02:
Was your pause command set as M25 before? The plugin intercepts that command and actually seems the pause mqtt command to the printer.

Kitki30, 2024-08-03:
idk, it was set to something default.

Kitki30, 2024-08-03:
M25 is set to sd cancel command

Kitki30, 2024-08-03:
[image]

jneilliii, 2024-08-03:
Yeah, so unless your printing files uploaded to OctoPrint directly into local storage this shouldn't effect how the plugin works. The plugin acts as virtual sd card to OctoPrint and the files that are printed should be seen as sd card files as indicated by the little icon in the upper right of the file.

Kitki30, 2024-08-03:
It was set to default: three commands m25 was the last

Kitki30, 2024-08-03:
changed it to only m25

Kitki30, 2024-08-03:
i cant see if its working im in my grandpa's home now
maybe i can see test it tomorrow

i mainly send prints through bambu studio, i use octoprint to see print progress from my wear os watch with octoapp but i see that octoprint have some cool plugins that makes printing better i host it on 8gb pi 5

Kitki30, 2024-08-03:
idk if its plugin problem or octoapp problem but if its possible can you fix that when i hit start printing button on octo app it shows me only files stored on rapsberry pi's sd card not printer sd card maybe its virtual sd card issue

Kitki30, 2024-08-03:
also can you fix that when the printer is not even turned on octo print thinks its operational (when you press connect button)
[image]

Kitki30, 2024-08-03:
the internet quality is poor here so i can reply late

jneilliii, 2024-08-03:
> idk if its plugin problem or octoapp problem but if its possible can you fix that when i hit start printing button on octo app it shows me only files stored on rapsberry pi's sd card not printer sd card maybe its virtual sd card issue

I have no idea how to filter OctoApp list of files, @crysxd may be able to tell us though.

jneilliii, 2024-08-03:
> also can you fix that when the printer is not even turned on octo print thinks its operational (when you press connect button) [...]

I will have to test this one out, prior to the recent refactor by @Valemos OctoPrint state would disconnect when the printer was powered off.

Kitki30, 2024-08-03:
ok

crysxd, 2024-08-03:
> I have no idea how to filter OctoApp list of files, @crysxd may be able to tell us though.

OctoApp doesn't support files from the printers and card. It's that simple 😅 

I never implemented it and it's not frequently requested

Kitki30, 2024-08-04:
> > > idk if its plugin problem or octoapp problem but if its possible can you fix that when i hit start printing button on octo app it shows me only files stored on rapsberry pi's sd card not printer sd card maybe its virtual sd card issue
> > 
> > 
> > I have no idea how to filter OctoApp list of files, @crysxd may be able to tell us though.
> 
> OctoApp doesn't support files from the printers SD card. It's that simple 😅
> 
> I never implemented it and it's not frequently requested

OK Thanks for reply!

Kitki30, 2024-08-04:
> OK Thanks for reply!

Does octoapp have wear os notifications or my watch is broken? I have octo companion plugin

crysxd, 2024-08-04:
That's not the place to discuss this, but notifications from the phone also show on the watch

Kitki30, 2024-08-05:
> That's not the place to discuss this, but notifications from the phone also show on the watch

ok

closed 2024-08-08 by Kitki30