jneilliii/OctoPrint-BambuPrinter #37: When bambu studio is turned on, octoprint sometimes doesnt update temperatures, etc.

Kitki30, 2024-08-03:
When bambu studio is turned on, octoprint sometimes doesnt update:

Temperatures and print status sd card file list also doesnt work.
My printer is bambu a1 mini

jneilliii, 2024-08-03:
This is as limitation of your printer's firmwarev from what I understand. It can only support a certain number of connections and only reports to the last connected "monitoring device". Are you cloud connected or local only, in both plugin settings and printer setup.

Kitki30, 2024-08-03:
lan only  mode is turned off but i use local connection

jneilliii, 2024-08-03:
> lan only mode is turned off but i use local connection

If Ian only mode is off on your printer then the plugin needs to be in cloud mode (use local access unchecked).

Kitki30, 2024-08-04:
ok, why does it need to be lan only mode on? Im gonna change it to cloud but my bambu account is google

Kitki30, 2024-08-04:
> > lan only mode is turned off but i use local connection
> 
> If Ian only mode is off on your printer then the plugin needs to be in cloud mode (use local access unchecked).

What do i need to type in region
Im in poland / europe

Kitki30, 2024-08-04:
> > > lan only mode is turned off but i use local connection
> > 
> > 
> > If Ian only mode is off on your printer then the plugin needs to be in cloud mode (use local access unchecked).
> 
> What do i need to type in region Im in poland / europe

ok typed europe and it works

closed 2024-08-08 by Kitki30