jneilliii/OctoPrint-BambuPrinter #38: Idle status shows as "Operational" breaking Tasmota-MQTT functionality

MikePhoseon, 2024-08-09:
Thanks for this great plugin.

Probably a feature request, any make it makes more sense to file this against Tasmota-MQTT as a feature request to choose the operating state that is looks for to trigger the Power Off option.

It appears whenever the printer is not printing and has entered the "idle" state, the reported state is "Operational", so unable to use the Power Off on Idle feature to turn off lights and fans after prints complete.

jneilliii, 2024-08-09:
If the plugin is not detecting a printing state from the printer than there's an issue with the connection in most cases. Most common is local only connection in the plugin settings, but a cloud connected printer. There are also some other fringe cases I've heard of like not having cache files option from cloud connected printers and a couple of others like MQTT connection limits imposed by Bambu firmware (dependent on model). 

To troubleshoot further, enable debug logging for the plugin in OctoPrint's plugin settings and reproduce the conditions and lets see what the logs show. Would be good to know your plugin settings too local vs cloud and model. I only have an X1C personally so troubleshooting can be difficult for me on other devices.

MikePhoseon, 2024-08-13:
I think it's actually detecting the state fine, I guess that the P1S might be reporting different states? The only states I've seen reported are "Operational" and "Printing from SD Card". I'll get some logs uploaded.

MikePhoseon, 2024-08-13:
[bambu_printer.log](https://github.com/user-attachments/files/16606150/bambu_printer.log)

[image]

Should have further specified this is a P1S in Cloud connected mode. 

After enabling logging I sent a job to the printer from OcraSlicer, and then cancelled it while the printer was heating up.

I was making the assumption that since the web interface was reporting "Operational" that was the state that tasmota_MQTT was seeing as well, and so it wasn't triggering. However looking at the log it looks like more likely the issue is that the state appears to just be jumping between FAILED, FINISH, and IDLE, so it's never actually staying on IDLE long enough for the timer to expire in tasmota_MQTT

MikePhoseon, 2024-08-24:
Took me a bit to actually reproduce the issue after enabling the logging for the Tasmota plugin, but was finally able to cause it by canceling a print (it does seem to be dependent on when the print is cancelled), in this case I'm just getting failure status messages:

`2024-08-23` 14:56:41,810 - octoprint.plugins.bambu_printer.BambuPrinter - DEBUG - Received printer state update: FAILED
`2024-08-23 14:56:41,814 - octoprint.plugins.bambu_printer.BambuPrinter - DEBUG - Received printer state update: FAILED

repeat, it hasn't changed in several hours and I haven't gotten any other status messages

I have the Tasmota plugin set to Debug in logging, and I rebooted the system, however the only messages I'm seeing from the plugin are:

```
2024-08-23 14:23:12,904 - octoprint.plugins.mqtt - DEBUG - Sent message: tasmota/plg-bambup1s/cmnd/POWER - , retain=True
2024-08-23 14:23:12,907 - octoprint.plugins.mqtt - DEBUG - Sent message: tasmota/lht-bambup1s/cmnd/POWER1 - , retain=True
2024-08-23 14:23:12,910 - octoprint.plugins.mqtt - DEBUG - Sent message: tasmota/fan-office/cmnd/POWER - , retain=True
2024-08-23 14:23:13,003 - octoprint.plugins.mqtt - DEBUG - Sent message: octoprint/plugin/tasmota - echo: OFF, retain=True
2024-08-23 14:23:13,018 - octoprint.plugins.mqtt - DEBUG - Sent message: octoprint/plugin/tasmota - echo: OFF, retain=True
2024-08-23 14:23:13,191 - octoprint.plugins.mqtt - DEBUG - Sent message: octoprint/plugin/tasmota - echo: ON, retain=True
2024-08-23 14:23:15,482 - octoprint.plugins.mqtt - DEBUG - Sent message: tasmota/lht-bambup1s/cmnd/POWER1 - ON, retain=True
2024-08-23 14:23:15,581 - octoprint.plugins.mqtt - DEBUG - Sent message: octoprint/plugin/tasmota - echo: ON, retain=True
2024-08-23 14:23:16,367 - octoprint.plugins.mqtt - DEBUG - Sent message: tasmota/fan-office/cmnd/POWER - ON, retain=True
2024-08-23 14:23:16,409 - octoprint.plugins.mqtt - DEBUG - Sent message: octoprint/plugin/tasmota - echo: ON, retain=True
2024-08-23 14:56:04,666 - octoprint.plugins.mqtt - DEBUG - Sent message: tasmota/plg-bambup1s/cmnd/POWER - , retain=True
2024-08-23 14:56:04,670 - octoprint.plugins.mqtt - DEBUG - Sent message: tasmota/lht-bambup1s/cmnd/POWER1 - , retain=True
2024-08-23 14:56:04,673 - octoprint.plugins.mqtt - DEBUG - Sent message: tasmota/fan-office/cmnd/POWER - , retain=True
2024-08-23 14:56:04,767 - octoprint.plugins.mqtt - DEBUG - Sent message: octoprint/plugin/tasmota - echo: ON, retain=True
2024-08-23 14:56:04,772 - octoprint.plugins.mqtt - DEBUG - Sent message: octoprint/plugin/tasmota - echo: ON, retain=True
2024-08-23 14:56:04,795 - octoprint.plugins.mqtt - DEBUG - Sent message: octoprint/plugin/tasmota - echo: ON, retain=True
```

So I'm not sure what's going on in that plugin, maybe I'm looking in the wrong log?

jneilliii, 2024-08-24:
Make sure to enable debug logging directly in the Tasmota plugin settings, and it will log into it's own separate log file, plugin_tasmota_debug.log.