jneilliii/OctoPrint-BambuPrinter #41: Obico not seeing printer is printing

clutchkick512, 2024-08-27:
So I have a great stream from the webcam connected to my pi, bambu plugin is connected with auth key for my cloud A1. But obico never detects that a print has started. Im sure im missing something simple here but im not sure what it is.

jneilliii, 2024-08-27:
In OctoPrint do you ever see the state panel change from operational to printing?

hashoom8, 2024-08-29:
the problem came back

jneilliii, 2024-08-29:
In isssue #22 you mentioned that restarting your network fixed the issue @hashoom8. In either case, I need to have you guys enable debug logging in OctoPrint's logging settings for the plugin, reproduce the issue, and then share your plugin_bambuprinter_debug.log file.

hashoom8, 2024-08-30:
please explain which one you want 
[image]
[image]

jneilliii, 2024-08-30:
share the plugin_bambu_printer_serial.log for starts.

aslabsalbeh, 2024-09-05:
Same here. The problem possibly started with the 0.1.3 update. Octoprint sees the print and shows the temp values etc.. but it doesn't show when a print is started. It stays on "operational".

Attached my log file hoping that it might help... Thanks!!

[plugin_bambu_printer_serial.log](https://github.com/user-attachments/files/16884139/plugin_bambu_printer_serial.log)

jneilliii, 2024-09-05:
from this log it appears to be getting into a file listing loop. can you also share octoprint.log there will be some additional information in there as well.

aslabsalbeh, 2024-09-05:
[octoprint.log](https://github.com/user-attachments/files/16896892/octoprint.log)

I started a print today and everything went fine after having the above issue in the last few days. The only thing I did was disconnect/reconnect the printer from Octoprint. That might have fixed the issue?

I attached the octoprint log anyways. Thank you for your support !

jneilliii, 2024-09-05:
That very well could have knocked something loose. I know the A1/P1 devices are more susceptible to stop sending info to connected clients and reconnecting basically tells the printer "send your updates to me". 

What happens when you send a print from the slicer the plugin realizes that the print starts and then checks to see if it is aware of the file's info (shows in file list). If it doesn't then it requests a file list from the printer to know what the file size is. It's that file size that is used to display print progress within OctoPrint.

ados8, 2025-01-12:
Having the same problem, brand new setup.
At first Obico wouldn't see any temps but after I switched to release candidate channel for Bambu plugin and switched to cloud it started pulling that data.
However, both Octoprint and Obico don't see it printing so Obico doesn't run the AI detection.