jneilliii/OctoPrint-BambuPrinter #42: Externally stopping/cancelling a print job not registering in OctoPrint

johnc2k, 2024-09-02:
* Octoprint: `1.10.2` (Docker Container)
* OctoPrint-BambuPrinter: `0.1.3`
* Bambu P1S: `01.06.01.00`

When stopping/cancelling a print job externally from OctoPrint (e.g. via Bambu Studio), OctoPrint does not register the print as stopped/cancelled and remains stuck on `Printing from SD`. 

Stopping/cancelling from Octoprint works fine.

[plugin_bambu_printer_serial.log](https://github.com/user-attachments/files/16841067/plugin_bambu_printer_serial.log)
[octoprint.log](https://github.com/user-attachments/files/16841071/octoprint.log)

jneilliii, 2024-09-02:
octoprint.log
> 2024-09-02 14:38:29,142 - octoprint.util.comm - INFO - Cancelling job on behalf of user admin
2024-09-02 14:38:29,142 - octoprint.util.comm - INFO - Changing monitoring state from "Printing from SD" to "Cancelling"
2024-09-02 14:38:29,143 - octoprint.printer.standard.job - INFO - Print job cancelled - origin: sdcard, path: 3DBenchy.gcode.3mf, owner: None, user: admin, fileposition: 179850, position: {'t': None, 'f': None, 'z': None, 'y': None, 'x': None, 'e': None}

I think it's picking up the cancel, but then never resets the messages being sent to OctoPrint back to Not SD Printing. 

>2024-09-02 14:38:32,067 - octoprint.plugins.bambu_printer.BambuPrinter - DEBUG - Received printer state update: FAILED

I should be able to know this needs resetting from this state update. 

BTW, I notice commands being sent after the cancel. You may want to remove any commands from OctoPrint's GCODE Scripts section to prevent any issues.

jneilliii, 2024-09-03:
I think I have this fixed in the above commit/branch. If you don't mind testing it out for me that would be greatly appreciated. Go into Plugin Manager > Get More, and at the bottom copy/paste this URL into ...from URL and click install. No need to uninstall first, but if not prompted to restart OctoPrint make sure to do that. 

```
https://github.com/jneilliii/OctoPrint-BambuPrinter/archive/refs/heads/bugfix/external_cancel.zip
```

johnc2k, 2024-09-03:
Unfortunately the same issue is still occurring 

[plugin_bambu_printer_serial.log](https://github.com/user-attachments/files/16846739/plugin_bambu_printer_serial.log)
[octoprint.log](https://github.com/user-attachments/files/16846740/octoprint.log)

jneilliii, 2024-09-04:
I just noticed something additional in both serial logs. The file being currently printed is not being picked up in the file list? Can you share how you are starting the file and if you are running cloud connected or not for the plugin and printer? 

```
2024-09-03 09:29:21,406 >>> Begin file list
2024-09-03 09:29:21,406 >>> articu~1.3mf 2032589 0x58bb8762 "Articulated_Slug-round_tail.gcode.3mf"
2024-09-03 09:29:21,417 >>> lead-s~1.3mf 2026748 0x59218709 "lead-screw-shield-v2.3mf"
2024-09-03 09:29:21,417 >>> End file list
...
2024-09-03 09:28:52,721 >>> File opened: 3DBenchy.gcode.3mf  Size: 1789634
2024-09-03 09:28:52,721 >>> File selected
...
2024-09-02 14:09:54,638 >>> Begin file list
2024-09-02 14:09:54,639 >>> articu~1.3mf 2032589 0x58bb8762 "Articulated_Slug-round_tail.gcode.3mf"
2024-09-02 14:09:54,652 >>> short_~2.3mf 21059949 0x5917aa91 "Short Grip.gcode.3mf"
2024-09-02 14:09:54,652 >>> End file list
...
2024-09-02 14:10:30,212 >>> File opened: 3DBenchy.gcode.3mf  Size: 2248129
2024-09-02 14:10:30,212 >>> File selected
```

johnc2k, 2024-09-04:
Ah yes thats my fault, I trimmed down the filename list for privacy

jneilliii, 2024-09-04:
ok, thanks that makes more sense. I think I actually was able to resolve this issue for real and have tested myself. If you follow the same steps as before to install with ..from URL field in Plugin Manager I think it should be working as expected again.