jneilliii/OctoPrint-BambuPrinter #44: How to use?

FredGenius, 2024-09-07:
I've installed the plugin, but I can't figure out what to do next - my printer is a P1S.

jneilliii, 2024-09-07:
Look in settings for the Bambu Printer section in plugins to setup the connection to your printer.

FredGenius, 2024-09-07:
Thanks, I just found it. Maybe add this too readme?

 

Fred Ö¿Ö

jneilliii, 2024-09-07:
I don't usually do documentation until I publish a plugin to the official repo. The fact you found this plugin means you are now part of the beta program...good luck.

jneilliii, 2024-09-07:
I recommend you read through past issues to understand some of the quirks related to the plugin.

FredGenius, 2024-09-07:
This is a game-changer! Will be my pleasure to help, thanks.

 

Fred Ö¿Ö

FredGenius, 2024-09-07:
Thanks, will do.

 

Fred Ö¿Ö

FredGenius, 2024-09-30:
I have the latest version of OctoPrint-BambuPrinter installed and I see a dramatic improvement – I can see files on the sd card and move the carriage, but when I tried to print a gcode file I got this error… 

And this status message…

As before, not complaining, hope this helps.

Thanks for your hard work, please let me know if there’s anything I can do to speed this along.

 

Fred Ö¿Ö

jneilliii, 2024-09-30:
sorry, screenshots don't come through on email replies @FredGenius so I have no way of seeing your messages.

FredGenius, 2024-09-30:
Apologies, error message reads ‘[bold] Suppressed command’, ‘The command T1000 was not sent to the printer:’ ‘[detail] Not queuing T1000, that tool doesn’t  exist according to the printer profile or was reported as invalid by the firmware. Make sure your printer profile is set up correctly.’

The State message reads ‘State: Offline after error Could not write to serial port’

 

Fred Ö¿Ö

jneilliii, 2024-09-30:
Yeah, that means you uploaded to local storage and not the SD card, or trying to start a print that is stored on the local storage of OctoPrint. The plugin only works as a virtual SD card, so you need to make sure to start the files that are indicated as being on the SD card. 

[image]

I personally filter the file list to only show the SD card files. 

[image]