jneilliii/OctoPrint-BambuPrinter #45: Is there a way to add ETA or Time left on Optiprint pls :P

tzx39, 2024-09-09:


tzx39, 2024-09-09:
Sorry I'm stupid I didn't what for it to show up sorry

jneilliii, 2024-09-10:
keep in mind the time estimate shown in OctoPrint may be far off of reality, it's a limitation of how the print progress is presented and the requirement of presenting the printer as a virtual sd card.

tzx39, 2024-09-10:
No it's really good maybe like 2 min it is a little legi and can we get the filament sensor cuz it just stops the print

jneilliii, 2024-09-10:
Not sure if I ever tested that scenario, but I believe there is no distinction between paused and cancelled with SD card printing, but I can definitely take a look. In the printer notification area of the side panel you should see the hms notification for filament runout though.

tzx39, 2024-09-10:
Can I ask how to get the thumbnails.
And good job on the plugin work good

jneilliii, 2024-09-12:
> Can I ask how to get the thumbnails. And good job on the plugin work good Thank you :)

splitting as separate feature request. the plugin currently doesn't support thumbnails. I'm unsure if I want to handle that here as part of this plugin, or integrate the functionality in my Slicer Thumbnail plugin.

closed 2024-10-10 by jneilliii