jneilliii/OctoPrint-BambuPrinter #49: No sd printing

tzx39, 2024-09-12:
What does that mean
Im printing samthing and it keeps spending this to octoprint 

Is this way it is not updating status

Recv: Not SD printing
Recv: Not SD printing
Recv: Not SD printing

jneilliii, 2024-09-12:
Based on this and your other issue #48 there seems to be disconnect between your printer and OctoPrint. Definitely need looks as described in that issue.

tzx39, 2024-09-13:
How do I connect the plugin? I tried everything, reset the printer, reset OctoPrint, generated new LAN codes, and nothing worked.
For now, I will use Bambu Handy and return to the plugin when I get a i 4, as so far everything was running on a  Pi Zero because I didn't have anything else. :)

jneilliii, 2024-09-13:
a pi zero should be sufficient, but again you have provided no details to try and help you. is your printer cloud connected or not. I assume it is because you mention bambu handy. What model Bambu Printer are you using? Because different models have different limitations. for example, I can run my X1C in cloud mode but still connect locally and A1 P1 devices I think don't allow this.

tzx39, 2024-09-13:
I apologize for not sending any details; I was a bit busy. I have the A1 Mini in LAN mode, but it's strange because it worked the first time and then stopped. Now I have a bit more time. Sorry :P

tzx39, 2024-10-01:
Dam still haven't figured it out, i use online mode for now

aslabsalbeh, 2024-10-23:
Same issue here. I'm also unable to connect the plugin through the cloud. When I click on login nothing happens. Logs show error 403 which might be indicating a connection problem of some sort?

Used to work fine few months ago, not sure what changed.

[octoprint.log](https://github.com/user-attachments/files/17484655/octoprint.log)
[plugin_bambu_printer_serial.log](https://github.com/user-attachments/files/17484656/plugin_bambu_printer_serial.log)

tzx39, 2024-10-23:
What host for the octoprint are you using

tzx39, 2024-10-23:
Did you reserve a ip for you printer

jneilliii, 2024-10-23:
All the people experiencing issues lately I suspect are related to issues with Bambu API and using the cloud connected printer and the plugin set to use cloud mode. For relevant discussion see the issue from the Home Assistant integration, which this plugin uses a module from (maintained in separate repo and not yet updated). 

https://github.com/greghesp/ha-bambulab/issues/612

aslabsalbeh, 2024-10-23:
Thank you @jneilliii 

I do have the printer connected to HA and never had issues since I set it up. It detects when a print job has started and shows bed temp etc... and I just saw the update of the addon on HA which presumably fixes some recent connection issues.

@tzx39 yes it has a static IP.

jneilliii, 2025-01-13:
you're printing SD card files and not trying to print a locally uploaded file into OctoPrint? you have to go into OctoPrint's settings, logging section and enable debug logging for octoprint.plugins.bambu and re-create the issue, then the octoprint.log and bambu_serial.log will have additional information in them.

hubortje, 2025-01-14:
I was not, as far as I was concerned, but I double-checked. Despite using “Upload to SD”, the files only got a checkmark icon, and the print didn't start, and froze Octopi in the “Cancelling” state. The issue consists, even after a restart.

Yet, the files are uploaded correctly to the printer, I can select and start printing them from the printer's display.

Octopi has trouble to report temperature and printing state, after a startup. I also cannot disconnect. Does this issue maybe correlate or is the main issue?
[plugin_bambu_printer_serial.log](https://github.com/user-attachments/files/18416404/plugin_bambu_printer_serial.log)

jneilliii, 2025-01-15:
> Octopi has trouble to report temperature and printing state, after a startup. I also cannot disconnect. Does this issue maybe correlate or is the main issue?

All of this correlates with a bad connection to the printer's data stream. Please share octoprint.log too.