jneilliii/OctoPrint-BambuPrinter #52: [BUG]: A1 not reporting printing status and temperature

Joelminer-satisfactory, 2024-10-08:
**Description**
When connected to my bambu A1, octoprint does not report the temperature correctly, it also doesn't show the status as printing, it just constantly repeats "not sd printing" in the terminal

**Debug Logs**
[octoprint.log](https://github.com/user-attachments/files/17297407/octoprint.log)
[plugin_bambu_printer_serial.log](https://github.com/user-attachments/files/17297408/plugin_bambu_printer_serial.log)

**Printer and Plugin Setting Details**
printer model: A1 + ams lite
box for local access: unchecked
printer is cloud connected, cloud connection works via bambu handy and bambu studio

jneilliii, 2024-10-08:
It doesn't appear that a file list is ever generated of files that have been sent to your printer from the slicer. I feel like this might be a connection issue between the plugin and cloud. I assume you were able to login and generate the token necessary for connecting the plugin to the cloud? I'm concerned this may be a connection limit issue possibly because looking at the log, it seems that the plugin was stuttering connection to MQTT on startup. What firmware version are you running on the printer because I think @synman mentioned recently that there was a firmware update that increased that limit I think.

Joelminer-satisfactory, 2024-10-08:
I'm using firmware 01.03.01.00

aslabsalbeh, 2024-10-28:
So Just to confirm, the connection problem is still persistent ?

I'm unable to have the plugin work locally or through the cloud. When clicking on "login" in the cloud connection settings, nothing happens and no tocken is created.

Locally, Octoprint shows the A1 as "operational" and that's it. Temps and print status are not being reported.

A1 firmware is up to date. Things were running normal a month or 2 ago..

jneilliii, 2024-10-28:
If you switch the release channel for the plugin to release candidate and update when prompted there are changes in that version that might resolve the cloud connectivity issues. I haven't had time to personally test it myself, hence it still being in release candidate state. If it's still an issue, please do provide full logs (with `octoprint.plugins.bambu_printer` set to debug in OctoPrint's logging section of settings).

aslabsalbeh, 2024-10-28:
Thank you. I installed the RC version but it's not showing in the settings menu.

jneilliii, 2024-10-28:
thanks, I can confirm the same. looks like I may have missed a file during release. in ocotprint.log do you have the error of missing filaments.json file?

aslabsalbeh, 2024-10-28:
Yes I do:

FileNotFoundError: [Errno 2] No such file or directory: '/home/orangepi/OctoPrint/lib/python3.9/site-packages/octoprint_bambu_printer/printer/pybambu/filaments.json'

jneilliii, 2024-10-28:
Ok, just fixed and verified. You can copy/paste this URL in plugin manager > get more > ,,,from URL to install and it should work.
```
https://github.com/jneilliii/OctoPrint-BambuPrinter/archive/refs/tags/0.1.8rc1.zip
```

aslabsalbeh, 2024-10-28:
I think this release fixed the issue. I'm now able to log-in with cloud connection and token was successfully created!

I'll keep you updated if I find any issues. Thank you you're a hero!!

aslabsalbeh, 2024-10-28:
After some testing, I can see that Octoprint is now correctly reporting the bed/hotend temperatures, however the issue related to print status is still present. Octoprint will show "State: Operational" even after the printer starts printing.

The attached log shows the same error as before "No SD Card printing"

[octoprint.log](https://github.com/user-attachments/files/17537266/octoprint.log)

jneilliii, 2024-10-28:
I think this might boil down to when printer is cloud connected, the connections to other services in local mode (ie file upload, listing, etc. via ftps) is blocked. From the log it seems that the printer never responds to the `M20` request for file list. If both printer and plugin are in local only mode I think this issue wouldn't exist, but would require you to test to validate that theory, it's just based on what I've heard in other discussions with other developers.

Joelminer-satisfactory, 2024-10-28:
I am not sure if this would also apply to the plugin, but I am able to access the files on the printer while its cloud connected via FTP using FileZilla, username being bblp and the password being the access code of the printer, see attached screenshot for the exact settings.
[image]

jneilliii, 2024-10-28:
Well, those are the settings the plugin uses to connect to ftps, so must be someone else causing the issue. Maybe that's was an old firmware version limitation that doesn't exist anymore. What firmware version is everyone running?

Joelminer-satisfactory, 2024-10-28:
running 01.03.01.02 on the printer, might be beta firmware but i'm not 100% as it doesn't really make it clear when beta firmware is being used

jneilliii, 2024-10-28:
I pushed version 0.1.8rc2 of the plugin after finding a couple of other spots where I needed to adjust for the transferring of the pybambu module into the plugin itself. Those would have only effected pause/cancel commands though and shouldn't have an impact on the file listing I don't think. 

Relative to the file listings though, like I mentioned, the list is pulled from the printer using [this ](https://github.com/jneilliii/OctoPrint-BambuPrinter/blob/rc/octoprint_bambu_printer/printer/file_system/ftps_client.py) patched ftp module in the plugin. It's one that I've found online in the bambu forums or part of some other integration. Maybe I need to tweak it more for A1 support.

Joelminer-satisfactory, 2024-10-28:
The file list generates fine, at least for me. The things not working are the temps (both always report at 21.3C with target being off), and it always saying not SD printing, even if it is printing.

jneilliii, 2024-10-28:
Are you running the 0.1.8rc2 version of the plugin? That should be fixed in that one.

Joelminer-satisfactory, 2024-10-28:
am running 0.1.8rc2, included the logs
[octoprint.log](https://github.com/user-attachments/files/17548455/octoprint.log)
[plugin_bambu_printer_serial.log](https://github.com/user-attachments/files/17548456/plugin_bambu_printer_serial.log)

jneilliii, 2024-10-28:
weird, it looks like it's just not establishing the connection to the cloud mqtt server `us.mqtt.bambulab.com` assuming it's not a China region. Have you tried going into settings and re-establishing the token by entering email and username and pressing login button?

cvsickle, 2024-10-29:
Just want to add to this discussion, as I'm having similar problems (I'm on [0.1.8rc3](https://github.com/jneilliii/OctoPrint-BambuPrinter/releases/tag/0.1.8rc3) now, but I was having the same issue on 0.1.7).

I'm in LAN only mode. Octoprint is getting temps, file info, AMS info, etc., but the State is stuck on "Operational" and doesn't change. To keep the logs short, here are my logs for a brief period of time. Please note that I connected and disconnected manually.

[serial.log](https://github.com/user-attachments/files/17549955/serial.log)
[plugin_bambu_printer_serial.log](https://github.com/user-attachments/files/17549956/plugin_bambu_printer_serial.log)

If there's anything I can do to help track this down, please let me know. Thanks for all your work on this plugin.

jneilliii, 2024-10-29:
The more data the better. A unaltered (you can prune it for any possible passwords, etc.) octoprint.log would be helpful and based on this snippet of serial.log were you printing "Alien_Cutie_0821003814.gcode.3mf" or "ub_out_4.gcode.3mf"?

cvsickle, 2024-10-29:
I was printing "Alien_Cutie_0821003814.gcode.3mf". (https://makerworld.com/en/models/644787#profileId-570879)

Here's my log:

[octoprint.log](https://github.com/user-attachments/files/17550065/octoprint.log)

jneilliii, 2024-10-29:
If you get a chance please enable debug logging for the whole plugin. I'm only really seeing printer disconnect messages in this octoprint.log. Once enabled you'll need to re-create the situation so I can get some more data in the log. 

[image]

cvsickle, 2024-10-29:
Not a problem. I enabled serial logging, enabled debug logging for the plugin, rebooted octoprint (with auto connection enabled on boot), let it run for a bit, and then manually disconnected before grabbing the logs.

[octoprint.log](https://github.com/user-attachments/files/17550214/octoprint.log)
[plugin_bambu_printer_serial.log](https://github.com/user-attachments/files/17550215/plugin_bambu_printer_serial.log)
[serial.log](https://github.com/user-attachments/files/17550216/serial.log)

aslabsalbeh, 2024-10-29:
Hi again!

I updated today to RC3 and I have no issues with the temperature reporting. RC2 fixed it and it's still working. I'm cloud connected, but I'm planning to try local connection only through Bambu Print.

RC3 appears to have fixed some issues. Octoprint is detecting the correct status of the printer and is showing "Printing" in stead of being stuck on "Operational".

However the moment the print order starts it immediately shows -100% and doesn't show the time left or print progress.

[image]

Octopapp is running and monitoring spagetti detection but showing -100%

Ocoteverwhere immediately sent an email when I started the print saying that it completed printing (100%)

Aside form this, my main concern which is spagetti detection is now running and Octoprint can detect print orders.

jneilliii, 2024-10-29:
> Edit: I can't believe I didn't mention this yet, but I'm using an X1C... not an A1.

well, the title of the issue was for A1, but it seems you are experiencing similar issues so could be related, but I think it might be a different issue altogether, rare condition with specific reasons why. 

I find this entry interesting...
> subtask_name='Cute Alien'

I need to look at that makerworld listing and look at the 3mf file, because there have been weird instances of named plates in a 3mf file not matching the file name resulting in stuff like this in your log. 

> No 3mf file found

When that happens it doesn't know the total size of the file in order to report to OctoPrint the percentage value, and that subtask_name attribute is what the plugin uses to figure that out and start reporting printing.

cvsickle, 2024-10-29:
Interesting... Now that you're saying that, the print state stopped working in 0.17 when I started printing that file. It had been working before. I'll get a different print going tomorrow and see if my issue goes away.

jneilliii, 2024-10-29:
If it works regularly with another file you might try downloading the stl and slicing it in the slicer and sending to the printer instead of using direct 3mf. might make a difference in getting the subtask_name to match the sliced file.

aslabsalbeh, 2024-10-29:
An update: Octoprint detected the print finish.

So the only issue is the progress % and Octoprint registering 100% progress when the print starts.

I'll print more models in the next few days and see how it behaves.

jneilliii, 2024-10-29:
> Interesting... Now that you're saying that, the print state stopped working in 0.17 when I started printing that file. It had been working before. I'll get a different print going tomorrow and see if my issue goes away.

realized  makerworld doesn't have stl file download option. I went ahead and extracted it and uploading here.

[Alien_Cutie.zip](https://github.com/user-attachments/files/17551357/Alien_Cutie.zip)

aslabsalbeh, 2024-10-29:
> realized makerworld doesn't have stl file download option. I went ahead and extracted it and uploading here.
>
> [Alien_Cutie.zip](https://github.com/user-attachments/files/17551357/Alien_Cutie.zip)

If you click the arrow next to "Open in Bambu Studio" a drop down menu will show an option to download STL files.

jneilliii, 2024-10-29:
That specific model though only has 3mf as option once that is clicked though.

aslabsalbeh, 2024-10-29:
> That specific model though only has 3mf as option once that is clicked though.

I opem the 3mf files in Bambu Studio & save them as STL when that option isn't available on makerworld.

jneilliii, 2024-10-29:
which is exactly what I did for the above...

jcostom, 2024-10-29:
Just chiming in here. My A1 is on 01.03.01.02, with AMS Lite on 00.00.07.94 (both show as latest version). I was stuck on v0.1.6 of this module because I wasn't getting any info while stuff was printing, must as @Joelminer-satisfactory reported with v0.1.7. Just rolled forward to v0.1.8rc3 and it seems to have resolved these issues for me.

Joelminer-satisfactory, 2024-10-29:
interesting to see you all are having no issues, am on rc3, but to no avail, still reporting 21.3 C and always operational with a resend ratio of 0% (idk if that matters)

Joelminer-satisfactory, 2024-10-29:
Okay I have done a reset of octoprint, and it now seems to work perfectly, I can start prints, it reports as printing, shows the files and I can see the temperatures.

aslabsalbeh, 2024-10-29:
> Okay I have done a reset of octoprint, and it now seems to work perfectly, I can start prints, it reports as printing, shows the files and I can see the temperatures.

1- Does it show print progress?
2- Are you connected through the cloud with a generated tocken?

Joelminer-satisfactory, 2024-10-30:
1. it sometimes does, but since rc4 it no longer seems to, it just sends sd printing byte 1/0, with the state being cancelling, and the progress being -100%
2. am connected to the cloud using the generated token

aslabsalbeh, 2024-10-30:
> 1. it sometimes does, but since rc4 it no longer seems to, it just sends sd printing byte 1/0, with the state being cancelling, and the progress being -100%
> 2. am connected to the cloud using the generated token

Yeah I'm having the same issues.

jneilliii, 2024-10-30:
This all boils back down to the way printing status is figured out. The plugin receives a notice of a file being printed based on subtask_name of the MQTT report topic's payload. If that filename does not match a file already retrieved in OctoPrint's file list (typical case when you send a file from Orca/BambuStudio), it attempts to relist the files on the printer via ftps so it knows the total size of the print, which then allows it to generate the printing x/total_bytes and give that to OctoPrint. Once OctoPrint sees both the filename and this printing x/total_bytes is when it should go into a "printing" state. I literally tested rc4 this morning with my local only connection to the printer and it worked exactly as described, went into printing state, printed, then when complete went back into Operational state.

Joelminer-satisfactory, 2024-10-30:
observation, trying to start a print through octoprint from a file uploaded to the sd card causes the following exception: "MicroSD Card read/write exception. please reinsert or replace MicroSD Card." When I uploaded from orcaslicer, it keeps reporting that its sd printing the last byte, while its not, the amount of bytes is also incorrect (553793 bytes on a 118kb file), files also show up weird in bambu studio so there may be something else going on with my printer here, ill have a look at it.

Joelminer-satisfactory, 2024-10-30:
interestingly when I start the print through orca slicer it always reports 100% of the file is done. When I use bambu handy to upload however, it reports sd printing byte 1/{total_bytes}. I believe this might be the difference between cloud connected and local, as orca slicer doesn't seem to use the cloud to connect to the printer, while bambu handy does.

jneilliii, 2024-11-01:
ok, I'm seeing the same when I started a print from the printer directly.

```
Recv: File opened: <file_name>.3mf  Size: 1344509
Recv: File selected
Recv: SD printing byte 1344509/1344509
```

it instantly jumped to 100%, which it may have rolled over from the previous completed print maybe. I'll dig in deeper this weekend.

aslabsalbeh, 2024-11-01:
> ok, I'm seeing the same when I started a print from the printer directly.
>
> ```
> Recv: File opened: <file_name>.3mf  Size: 1344509
> Recv: File selected
> Recv: SD printing byte 1344509/1344509
> ```
>
> it [...]

I started a print from Bambu Studio yesterday and had the 100% reported. Later on I wanted to print the same model, so started it from the A1 itself - from the files saved on SD card. This time it started from 0% and all went well.

jneilliii, 2024-11-01:
When it's not working as expected are you guys seeing this message in octoprint.log?

> octoprint.filemanager.NoSuchStorage: No storage configured for destination sdcard

jneilliii, 2024-11-01:
and messages like

> 2024-11-01 17:38:18,669 - octoprint.plugins.bambu_printer.BambuPrinter.states - DEBUG - No 3mf file found for PrintJob(print_percentage=0, gcode_state='RUNNING', file_type_icon='mdi:file', gcode_file='/data/Metadata/plate_1.gcode', subtask_name='Kaiser Clip 2.0 PLA.gcode.3mf', start_time=None, end_time=None, remaining_time=51, current_layer=0, total_layers=95, print_error=0, print_weight=0, print_length=0, print_bed_type='unknown', print_type='local', _ams_print_weights=[0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], _ams_print_lengths=[0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0])

which ends up causing a file list refresh loop, constant `Begin file list`...`End file list` messages in the terminal tab of OctoPrint.

jneilliii, 2024-11-01:
ok...another point of potential reasoning why this is happening. Does the files that aren't being detected as starting have additional periods in the file name? I think the issue might be related to how the filenames are getting filtered out. These two definitely don't work.

```
sexy_lady_PLA_0.1mm.3mf
Kaiser Clip 2.0 PLA.gcode.3mf
```

and I think it's because of the extra periods in file name that aren't `.gcode.3mf`. For those files the suffixes for the file are determined to be `1mm, 3mf` and `0 PLA, gcode, 3mf`, and the plugin does try to get the data based on those suffixes matching `gcode, 3mf`, so the extra ones are breaking the "all" comparison.

jneilliii, 2024-11-02:
So I released version 0.1.8rc5 that will resolve the issue of files with additional periods in the filename, but there is still something a little off I think. Current printing file is picked up on first connection to printer, but then the next print seems to go into that instant 100% printing status.

Joelminer-satisfactory, 2024-11-02:
Updated to rc5, I can now see progress being updated every so often (same frequency as in bambu studio/handy), do note I did switch to a local connection, but the printer is still cloud connected.

Joelminer-satisfactory, 2024-11-02:
interestingly when I start the print from the bambu handy app it doesn't seem to work, staying at 100% permanently.

Joelminer-satisfactory, 2024-11-02:
When I print a file from the sd card, it says the following:
`2024-11-02 16:55:24,328 - octoprint.plugins.bambu_printer.BambuPrinter.states - DEBUG - No 3mf file found for PrintJob(print_percentage=91, gcode_state='RUNNING', file_type_icon='mdi:file', gcode_file='Battle Frog Leapster 3-in-1.3mf', subtask_name='PTFE & Cable Guide for A1 series', start_time=datetime.datetime(2024, 11, 2, 16, 28), end_time=datetime.datetime(2024, 11, 2, 16, 57), remaining_time=3, current_layer=82, total_layers=96, print_error=0, print_weight=5.8, print_length=1.95, print_bed_type='High Temp Plate', print_type='local', _ams_print_weights=[0, 0, 0, 5.8, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], _ams_print_lengths=[0, 0, 0, 1.9499999999999997, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0])`
However, when the file list is generated, the 3mf file is there, see the lines below:
`2024-11-02 16:55:24,329 - octoprint.plugins.bambu_printer.BambuPrinter.serial - DEBUG - >>> Begin file list
2024-11-02 16:55:24,333 - octoprint.plugins.bambu_printer.BambuPrinter.serial - DEBUG - >>> battle~1.3mf 1004753 0x5962be73 "Battle Frog Leapster 3-in-1.3mf"
2024-11-02 16:55:24,337 - octoprint.plugins.bambu_printer.BambuPrinter.serial - DEBUG - >>> End file list`
This all results in octoprint not showing progress/listing the printer as printing.

jneilliii, 2024-11-02:
this is what I was still working on. the issue is that 
`subtask_name='PTFE & Cable Guide for A1 series'` 
does not equal 
`"Battle Frog Leapster 3-in-1.3mf"`

`gcode_file='Battle Frog Leapster 3-in-1.3mf'`
does match the file name in the returned file list, however on other devices or possibly it's local only mode of the printer gcode_file returns something completely different. for example, I've been testing with just a small 10x10x2 cube and the gcode_file parameter of job info returns `/data/Metadata/plate_1.gcode`, so that's unusable. 

so as you can see I'm struggling with all kinds of random problems because of inconsistency between how the slicer(s), handy app, and local printer starting submit the print job and the subtask_name not being the same.. 

any idea why that print would report a subtask_name`PTFE & Cable Guide for A1 series`?

Joelminer-satisfactory, 2024-11-02:
> any idea why that print would report a subtask_namePTFE & Cable Guide for A1 series?
I don't have it open anymore, but I think that's the project name in bambu studio (or in my case orca slicer), I printed a makerworld file with that name before the battle frog, and instead of creating a new project I just deleted the original model, and imported the new one.

jneilliii, 2024-11-02:
> instead of creating a new project I just deleted the original model, and imported the new one.

This is what I was thinking might have been the case, which is one of the quirks that are hard to be programmed around, especially with the difference with `gcode_file` between the different printer models.

Joelminer-satisfactory, 2024-11-02:
one option might be to see if either `gcode_file` or `subtask_name` exists? As it appears to me, at least one of them will have the actual file name.

jneilliii, 2024-11-02:
Yeah, was considering that as an option, but adding additional lookup round trips through the file list doesn't feel efficient. I'm already thinking that there may be a bunch of unneeded lookups currently in the codebase, which was something I missed when reviewing a big refactor of the plugin by someone else. I will consider that as an option though now that I'm looking at this file lookup code more in depth.

jneilliii, 2024-11-03:
If I send a file from Orca everything works fine. If I start a print from the printer or OctoPrint it seems to get stuck on 1/x bytes printing. As a result, when the print does finish it never triggers printing all the file and is seen as cancelled by OctoPrint. I can't tell yet what variable isn't getting reset during the print start or finish process that would cause this weird condition. 

file started from printer...
```
2024-11-03 11:03:24,402 - octoprint.plugins.bambu_printer.BambuPrinter - DEBUG - Received printer state update: FINISH
2024-11-03 11:03:24,482 - octoprint.plugins.bambu_printer.BambuPrinter.serial - DEBUG - >>> Not SD printing
2024-11-03 11:03:25,493 - octoprint.plugins.bambu_printer.BambuPrinter.serial - DEBUG - >>> Not SD printing
2024-11-03 11:03:25,606 - octoprint.plugins.bambu_printer.BambuPrinter - DEBUG - Received printer state update: RUNNING
2024-11-03 11:03:25,607 - octoprint.plugins.bambu_printer.BambuPrinter - DEBUG - Changing state from IdleState to PrintingState
2024-11-03 11:03:25,608 - octoprint.plugins.bambu_printer.BambuPrinter - DEBUG - Select project file: Test.3.gcode.3mf
2024-11-03 11:03:25,611 - octoprint.plugins.bambu_printer.BambuPrinter - DEBUG - File already selected: Test.3.gcode.3mf
2024-11-03 11:03:25,611 - octoprint.plugins.bambu_printer.BambuPrinter - DEBUG - SD File Print finishing: Test.3.gcode.3mf
2024-11-03 11:03:25,611 - octoprint.plugins.bambu_printer.BambuPrinter - DEBUG - Changing state from PrintingState to IdleState
2024-11-03 11:03:25,610 - octoprint.plugins.bambu_printer.BambuPrinter.serial - DEBUG - >>> File opened: Test.3.gcode.3mf  Size: 45391
2024-11-03 11:03:25,614 - octoprint.plugins.bambu_printer.BambuPrinter.serial - DEBUG - >>> File selected
2024-11-03 11:03:25,616 - octoprint.printer.standard.job - INFO - Print job selected - origin: sdcard, path: Test.3.gcode.3mf, owner: None, user: None
2024-11-03 11:03:25,623 - octoprint.plugins.bambu_printer.BambuPrinter.serial - DEBUG - >>> SD printing byte 45391/45391
2024-11-03 11:03:25,631 - octoprint.plugins.bambu_printer.BambuPrinter.serial - DEBUG - >>> Done printing file2024-11-03 11:03:25,633 - octoprint.plugins.bambu_printer.BambuPrinter.serial - DEBUG - >>> Not SD printing
2024-11-03 11:03:26,319 - octoprint.plugins.bambu_printer.BambuPrinter.serial - DEBUG - >>> T:136.00/ 0.00 B:34.00/ 0.00 C:30.00/ 0.00 @:64
2024-11-03 11:03:26,359 - octoprint.plugins.bambu_printer.BambuPrinter - DEBUG - Received printer state update: RUNNING
2024-11-03 11:03:26,360 - octoprint.plugins.bambu_printer.BambuPrinter - DEBUG - Changing state from IdleState to PrintingState
2024-11-03 11:03:26,360 - octoprint.plugins.bambu_printer.BambuPrinter - DEBUG - Select project file: Test.3.gcode.3mf
2024-11-03 11:03:26,361 - octoprint.plugins.bambu_printer.BambuPrinter.serial - DEBUG - >>> File opened: Test.3.gcode.3mf  Size: 45391
2024-11-03 11:03:26,362 - octoprint.plugins.bambu_printer.BambuPrinter.serial - DEBUG - >>> File selected
2024-11-03 11:03:26,365 - octoprint.printer.standard.job - INFO - Print job selected - origin: sdcard, path: Test.3.gcode.3mf, owner: None, user: None
2024-11-03 11:03:26,505 - octoprint.plugins.bambu_printer.BambuPrinter.serial - DEBUG - >>> SD printing byte 1/45391
```

Joelminer-satisfactory, 2024-11-04:
interesting, I haven't yet tried starting from the printer, but when I start a print from octoprint, I get an SD card error if its a file from the printer's SD, or it sends batches of commands really quickly, which causes the printer to get confused if the file is on the raspberry pi's SD card.

jneilliii, 2024-11-04:
I get some errors occasionally when starting from OctoPrint but the print still starts. Would be interested in seeing what error you are referring to.

[35 more replies]