jneilliii/OctoPrint-BambuPrinter #54: [BUG]: Plugin connected even if printer is offline

jneilliii, 2024-10-26:
**Describe the Bug**
With printer powered off the plugin will still report connection, temperature data, and sd printing status.

**Expected Behavior**
To not connect in OctoPrint connection panel. 

**Debug Logs**
[plugin_bambu_printer_serial.log](https://github.com/user-attachments/files/17528520/plugin_bambu_printer_serial.log)
[octoprint.log](https://github.com/user-attachments/files/17528521/octoprint.log)

**Screenshots**

**Printer and Plugin Setting Details**

* Printer model? X1C
* Is your printer connected to Bambu Cloud? No
* Is the  plugin configured for local access only? Yes

benjmharless, 2025-03-22:
I have a P1S and it does the same thing

jneilliii, 2025-03-22:
are you running the rc version?