jneilliii/OctoPrint-BambuPrinter #57: [BUG]: Access code required when connecting to Bambu Cloud

Nicolas-Chasteler, 2024-10-28:
**Describe the Bug**
Version 0.1.8rc3
Connecting to printer using Bambu Cloud requires a value in the access code field. The virtual printer doesn't initialize and a connection can't be made and a "No serial port found" error is displayed. 

This can be fixed by adding any value in the access code field.

**Expected Behavior**
Access code is not required to be filled in to connect through the Bambu Cloud.

**Debug Logs**
N/A

**Screenshots**
No access code:
[image]
[image]

Access code:
[image]
[image]

**Printer and Plugin Setting Details**

* Printer model? P1S
* Is your printer connected to Bambu Cloud? Yes
* Is the  plugin configured for local access only? No

jneilliii, 2024-10-28:
I suspect this is just an internal issue with creating the bambu client connection. The module itself requires this as a parameter, but I will do some internal testing/code review to validate.

jneilliii, 2024-10-29:
Ok, it didn't take me long to figure out that the access code is still required in order to pull file listings and access to timelapses from the printer via ftps. I don't think there's any other way to get that directly from the cloud mqtt connection, which is what temperature, print progress, etc. data comes from.

closed 2024-10-29 by jneilliii