jneilliii/OctoPrint-BambuPrinter #59: [BUG]: unable to connect to the cloud service on A1 printer

penguinpatrol, 2024-11-05:
**Describe the Bug**
unable to connect to the cloud service on A1 printer i can enter in the login  details but the login button doesn't work (there is no response and the auth token does not fill)

**Expected Behavior**
i expect to be able to log in to the cloud service

**Debug Logs**
 
[plugin_bambu_printer_serial.log](https://github.com/user-attachments/files/17635637/plugin_bambu_printer_serial.log)

**Screenshots**

**Printer and Plugin Setting Details**

* Printer model? A1
* Is your printer connected to Bambu Cloud? yes
* Is the  plugin configured for local access only? it does work locally but i wish to connect it through the cloud service

jneilliii, 2024-11-05:
This is an issue created by Bambu and their new workflow for authentication. It's going to be a while before I implement those changes in the plugin.

AJolly, 2024-11-09:
@penguinpatrol if you use HA and install that plugin you can then pull the correct auth token from core.config_entries

penguinpatrol, 2024-11-09:
oh awesome i do use that where will i find the config entry??

jneilliii, 2024-11-09:
I've started working on the authentication flow within the plugin's settings. Will be available soon once I've been able to test it.

penguinpatrol, 2024-11-09:
i managed to find it but nothing seems to have changed truoing to enter it so hopefully the propper fix will sort it when its released

jneilliii, 2024-11-10:
I've just released version 0.1.8rc8 that has an updated login flow for the new verification processes required by Bambu. Enter email/password and press login, then using either 2fa verification code or verification code emailed to the given email address and click verify. If all goes smooth the auth token will be populated and you just click save.

penguinpatrol, 2024-11-10:
how would i update this? the software update tab shows the plugin as being up to date?

penguinpatrol, 2024-11-10:
oh thanks! i am currently installing from the zip

penguinpatrol, 2024-11-10:
got it all connected it seems to be working mostly fine except the prints i hav e done so far (sent through bambu studio) dont seem to register in octoprint (i also cant see any of the files on the sd card in octoprint)
[image]
[image]

its registering the ams lite and temp sensors fine just it doest seem to be detecting any active prints

jneilliii, 2024-11-10:
in terminal tab do you see multiple file list refreshes over and over? when sending files through the cloud the option to cache printing files on the printer is a requirement. make sure that option is enabled on your printer.

penguinpatrol, 2024-11-10:
[image]
this is what's looping, im not sure if the caching option is enabled so ill make sure it is in a bit!

penguinpatrol, 2024-11-10:
althought i cant seem to find the setting? is it on the printer itself? does it not show while printing?

jneilliii, 2024-11-10:
I'm OctoPrint's settings enable debug for octoprint.plugins.bambu_printer in the logging section, restart OctoPrint and let's get some detailed logs to troubleshoot. I am assuming you have an SD card in the printer.

penguinpatrol, 2024-11-10:
yeah sd card in the printer and print is currently running

penguinpatrol, 2024-11-10:
[plugin_bambu_printer_serial (2).log](https://github.com/user-attachments/files/17692354/plugin_bambu_printer_serial.2.log)
i think this is the correct thing?

jneilliii, 2024-11-10:
perfect...

> 2024-11-10 17:21:42,342 ftplib.error_perm: 530

this seems to be a permissions error when trying to retrieve the file list over ftps. you do also have the access code entered in the plugin's settings? 

it's possible this is a limitation of A1/P1 devices. I have heard from the development community that once cloud connected some services are no longer available locally to connect to.

penguinpatrol, 2024-11-10:
yeah i have the access code in i know it works locally as i tried that before its just the cloud not registering the current print, i know home assistant bambu connection work fine all of thoes scanners and such are connected.

penguinpatrol, 2024-11-10:
ah wait i did change the wifi a few hours ago that may jhave something to do with it

jneilliii, 2024-11-10:
the file list will be generated locally over ftps direct connection to printer, so the IP address and access code must be present, then when the file list is able to be generated the printing status from cloud should be picked up properly.

penguinpatrol, 2024-11-10:
[image]
its all in and i am pretty sure all correct, i can double check in a few hours when the print is done just to make sure but i dont belive it has changed

lordCONAN, 2024-12-01:
Using Version 0.1.8rc11 after entering the verification code from the email, and click verify, the auth token isn't generated, and another verification code is sent from bambu labs.

jneilliii, 2024-12-01:
any errors in logs? could you reproduce the process with debug logging enabled, which will include some additional information for assistance troubleshooting the process. can't guarantee there will be anything obvious.

lordCONAN, 2024-12-01:
Got this in the logs:

2024-12-01 13:51:58,895 - octoprint.server.api - ERROR - Error while executing SimpleApiPlugin bambu_printer
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/server/api/__init__.py", line 161, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/util/__init__.py", line 1686, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint_bambu_printer/bambu_print_plugin.py", line 151, in on_api_command
    auth_response = self._bambu_cloud.login_with_verification_code(data["password"])
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint_bambu_printer/printer/pybambu/bambu_cloud.py", line 241, in login_with_verification_code
    self._username = self._get_username_from_authentication_token()
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint_bambu_printer/printer/pybambu/bambu_cloud.py", line 163, in _get_username_from_authentication_token
    b64_string = self._auth_token.split(".")[1]
IndexError: list index out of range
2024-12-01 13:53:10,568 - octoprint.plugins.bambu_printer - INFO - Verifying user xxx.xxx@gmail.com
2024-12-01 13:53:21,725 - octoprint.plugins.bambu_printer - INFO - Error verifying: codeExpired

jneilliii, 2024-12-01:
thanks, I'll have to do some additional testing and check upstream with the pybambu module to see if something changed in the authentication flow again. I vaguely remember having a double email once, but I think this error indicates an issue with getting the underlying username.

jneilliii, 2024-12-01:
Just released a potential fix in https://github.com/jneilliii/OctoPrint-BambuPrinter/releases/tag/0.1.8rc12, but I haven't personally tested it yet. I have merged in the upstream changes for pybambu and adjusted to get it working in local only mode the way I use it. Give it a try and see if your authentication with cloud is working now.

lordCONAN, 2024-12-01:
Just downloaded it, and the plugin is no longer able to load

2024-12-02 06:37:24,977 - octoprint.plugin.core - ERROR - Error loading plugin bambu_printer
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/plugin/core.py", line 1297, in _import_plugin
    module = _load_module(module_name, spec)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/plugin/core.py", line 52, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/vendor/imp.py", line 238, in load_module
    return load_package(name, filename)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/vendor/imp.py", line 212, in load_package
    return _load(spec)
  File "<frozen importlib._bootstrap>", line 711, in _load
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint_bambu_printer/__init__.py", line 6, in <module>
    from .bambu_print_plugin import BambuPrintPlugin
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint_bambu_printer/bambu_print_plugin.py", line 29, in <module>
    from octoprint_bambu_printer.printer.pybambu import BambuCloud
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint_bambu_printer/printer/pybambu/__init__.py", line 3, in <module>
    from .bambu_client import BambuClient
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint_bambu_printer/printer/pybambu/bambu_client.py", line 19, in <module>
    from .bambu_cloud import BambuCloud
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint_bambu_printer/printer/pybambu/bambu_cloud.py", line 28, in <module>
    from .const import (
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint_bambu_printer/printer/pybambu/const.py", line 10, in <module>
    from .const_hms_errors import HMS_ERRORS
ModuleNotFoundError: No module named 'octoprint_bambu_printer.printer.pybambu.const_hms_errors'

jneilliii, 2024-12-01:
Hopefully in fixed in 0.1.8rc13 I just released.

lordCONAN, 2024-12-01:
The addon loads now, but when I enter my password and click login a verification code is sent, but there is no longer a textbox to enter the verification code, and the following error was in the log:

2024-12-02 07:24:44,455 - octoprint.server.api - ERROR - Error while executing SimpleApiPlugin bambu_printer
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/server/api/__init__.py", line 161, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/util/__init__.py", line 1686, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint_bambu_printer/bambu_print_plugin.py", line 136, in on_api_command
    auth_response = self._bambu_cloud.login(data["region"], data["email"], data["password"])
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint_bambu_printer/printer/pybambu/bambu_cloud.py", line 360, in login
    result = self._get_authentication_token()
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint_bambu_printer/printer/pybambu/bambu_cloud.py", line 195, in _get_authentication_token
    raise EmailCodeRequiredError()
octoprint_bambu_printer.printer.pybambu.bambu_cloud.EmailCodeRequiredError: Email code required

jneilliii, 2024-12-01:
Ok, this will require me to try the process myself and step through it. I was hoping for more of a drop in replacement from the ha project, but they've changed quite a bit. I'll get to it eventually this week.

jneilliii, 2024-12-07:
Just released version 0.1.8rc14, which I was able to use to login to cloud and verify the code emailed to me.

lordCONAN, 2024-12-07:
I was able to login and verify the code and the auth token field was populated, but it seems to provide no method to connect to the printer in Octoprint via the cloud connection's MQTT.

jneilliii, 2024-12-07:
Once those are saved and you disconnect and connect back to printer in connection panel, it automatically will connect to cloud mqtt.

lordCONAN, 2024-12-08:
I didn't realise I needed to enter the local info as well for it to load, but once I had it entered it connected. The only problem is that once connected it seems to not work. This turns up in the log:

2024-12-08 07:50:45,854 - octoprint.plugins.bambu_printer.BambuPrinter.serial - DEBUG - ------------------------------------------------------------------------------
2024-12-08 07:50:45,996 - octoprint.plugins.bambu_printer.BambuPrinter - DEBUG - connecting via local mqtt: False
2024-12-08 07:50:46,036 - octoprint.plugins.bambu_printer.BambuPrinter - DEBUG - on disconnect called
2024-12-08 07:50:46,037 - octoprint.plugins.bambu_printer.BambuPrinter - DEBUG - on connect called
2024-12-08 07:50:46,101 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - bambu connection status: False
2024-12-08 07:50:46,102 - octoprint.util.comm - INFO - Serial detection: Handshake attempt #1 with timeout 2.0s
2024-12-08 07:50:46,102 - octoprint.plugins.bambu_printer.BambuPrinter - DEBUG - Setting read timeout to 2.0s
2024-12-08 07:50:46,103 - octoprint.plugins.bambu_printer.BambuPrinter.serial - DEBUG - <<< 

2024-12-08 07:50:46,135 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2024-12-08 07:50:46,164 - octoprint.plugins.bambu_printer.BambuPrinter.serial - DEBUG - >>> Printer connection complete
2024-12-08 07:50:46,172 - octoprint.plugins.bambu_printer.BambuPrinter.serial - DEBUG - <<< N0 M110 N0*125

2024-12-08 07:50:46,224 - octoprint.plugins.bambu_printer.BambuPrinter.serial - DEBUG - >>> ok
2024-12-08 07:50:46,262 - octoprint.plugins.bambu_printer.BambuPrinter - DEBUG - Setting read timeout to 2.0s
2024-12-08 07:50:46,331 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial connection" to "Operational"
2024-12-08 07:50:46,344 - octoprint.plugins.bambu_printer.BambuPrinter.serial - WARNING - Not a valid gcode command ""
2024-12-08 07:50:46,362 - octoprint.server - INFO - Autorefresh of serial port list stopped
2024-12-08 07:50:46,377 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2024-12-08 07:50:46,376 - octoprint.plugins.bambu_printer.BambuPrinter.serial - DEBUG - >>> ok
2024-12-08 07:50:46,404 - octoprint.plugins.bambu_printer.BambuPrinter.serial - DEBUG - <<< N0 M110 N0*125

2024-12-08 07:50:46,464 - octoprint.plugins.bambu_printer.BambuPrinter.serial - DEBUG - <<< N1 M115*39

2024-12-08 07:50:46,485 - octoprint.plugins.bambu_printer.BambuPrinter.serial - DEBUG - >>> ok
2024-12-08 07:50:46,533 - octoprint.plugins.bambu_printer.BambuPrinter - DEBUG - processing gcode M115 command = M115
2024-12-08 07:50:46,578 - octoprint.plugins.bambu_printer.BambuPrinter.serial - DEBUG - <<< N2 M21*18

2024-12-08 07:50:46,592 - octoprint.plugins.bambu_printer.BambuPrinter.gcode_executor - DEBUG - Executing M115 (Firmware Info)
2024-12-08 07:50:51,613 - octoprint_bambu_printer.printer.pybambu - WARNING - On Disconnect: Printer disconnected with error code: 16
2024-12-08 07:50:58,082 - octoprint_bambu_printer.printer.pybambu - WARNING - On Disconnect: Printer disconnected with error code: 16
2024-12-08 07:51:04,592 - octoprint_bambu_printer.printer.pybambu - WARNING - On Disconnect: Printer disconnected with error code: 16
2024-12-08 07:51:10,998 - octoprint_bambu_printer.printer.pybambu - WARNING - On Disconnect: Printer disconnected with error code: 16
2024-12-08 07:51:16,587 - octoprint.util.comm - INFO - Communication timeout while idle, trying to trigger response from printer.
2024-12-08 07:51:16,598 - octoprint.plugins.bambu_printer.BambuPrinter.serial - DEBUG - <<< N3 M105*36

2024-12-08 07:51:17,392 - octoprint_bambu_printer.printer.pybambu - WARNING - On Disconnect: Printer disconnected with error code: 16
2024-12-08 07:51:18,592 - octoprint.util.comm - INFO - Communication timeout while idle, trying to trigger response from printer.
2024-12-08 07:51:20,599 - octoprint.util.comm - INFO - No response from printer after 3 consecutive communication timeouts, considering it dead.
2024-12-08 07:51:23,852 - octoprint_bambu_printer.printer.pybambu - WARNING - On Disconnect: Printer disconnected with error code: 16
2024-12-08 07:51:26,614 - octoprint.plugins.tracking - INFO - Sent tracking event commerror_timeout, payload: {'commerror_text': 'Too many consecutive timeouts, printer still connected and alive?'}
2024-12-08 07:51:30,298 - octoprint_bambu_printer.printer.pybambu - WARNING - On Disconnect: Printer disconnected with error code: 16
2024-12-08 07:51:36,709 - octoprint_bambu_printer.printer.pybambu - WARNING - On Disconnect: Printer disconnected with error code: 16

jneilliii, 2024-12-08:
hmm, seems error code 16 might be a problem with handling a bunch of messages at once after connection, or something else is getting in the way. I was worried that even after fixing the login issues that this might be a problem, because I had heard that A1/P1 series printers block the local file access via ftps that the plugin uses to retrieve the file list. I'll have to look into the pybambu module code to see if there is anything in there to help with listing/uploading files while in cloud connected mode. Without the file listing being possible, the plugin has no way of figuring out what is going on with ongoing prints, etc.

lordCONAN, 2024-12-08:
I guess there's a fundamental difference in the way octoprint and home assistant handle it. The HA component has no problem showing on-going prints, but I don't think it ever retrieves a file list.

jneilliii, 2024-12-08:
Yes, very much so. The HA integration doesn't care about or have any knowledge of the files stored on the printer, but OctoPrint needs that information to calculate the print progress based on file size. It's very convoluted, but is the only way that the integration can work with OctoPrint, using the known filename that is displayed in file list of OctoPrint and it's file size to create a printing x/y bytes as an emulated SD card print.

lordCONAN, 2024-12-08:
If there's anything I can do to help testing it, let me know. I assume it must be possible to get the file list somehow, as bambu studio is able to do it.

jneilliii, 2024-12-08:
That's my assumption as well, assuming there's an API file listing approach.

lordCONAN, 2024-12-16:
Seeing as it's required to have the ip, serial number and access code for the plugin to work anyway, could you just use those to get the file list via ftp?

lordCONAN, 2024-12-16:
I just checked with my A1 on the latest firmware (01.04.00.00) and I'm able to use WinSCP to ftp with TLS into my printer and view the files on (I presume) the SD card while the printer is connected to the cloud.

jneilliii, 2024-12-16:
> I just checked with my A1 on the latest firmware (01.04.00.00) and I'm able to use WinSCP to ftp with TLS into my printer and view the files on (I presume) the SD card while the printer is connected to the cloud.

interesting, because ftps access to the printer is exactly how the plugin attempts to load the file list. In that view do you see a cache folder? What's inside that cache folder?

lordCONAN, 2024-12-16:
Top level is

cache
model
timelapse
recorder
logger
ipcam
image

Then inside the cache folder is a all the gcode, .bbl and .3mf files for stuff that I've printed.

On a side note, when I first tried to connect to the printer via ftps, it timed out, but after I'd opened bambu studio to make sure the printer was online, I was then able to connect via ftps.