jneilliii/OctoPrint-BambuPrinter #61: [BUG]: Unable to connect to cloud service on X1C

singhm29, 2024-12-01:
**Describe the Bug**
Unable to connect to the cloud service on an X1C after getting verification email for 2FA

**Expected Behavior**
To recieve an authentication token after entering the verficiation code

**Debug Logs**
[plugin_bambu_printer_serial.log](https://github.com/user-attachments/files/17967506/plugin_bambu_printer_serial.log)
[serial.log](https://github.com/user-attachments/files/17967508/serial.log)

Please let me know if the above isnt adequate. First time posting logs

**Screenshots**

**Printer and Plugin Setting Details**

* Printer model? X1C
* Is your printer connected to Bambu Cloud? Yes
* Is the  plugin configured for local access only? No and it seems to not work locally either. Prior to the 2FA introduction I was using this plugin sucessfully!

I have read the a similar issue with the A1 in the other thread and looked at the approaches between that and the A1 not reporting printing status and temperature thread. At least a month a go I was actively using the plugin but after some issues not being able to actually issue commands through the terminal I thought maybe my authenticaiton token was stale with the new 2FA and decided to reinstall and follow steps to install the releaese candidate. Unfortunately now I cant get the auth token to move forward.

jneilliii, 2024-12-01:
There have been updates in the release candidate that have the new 2fa verification process incorporated. Go into software update settings and change the plugin's release channel to release candidate and update when prompted.

singhm29, 2024-12-01:
Thanks for the quick response. To clarify I am using Bambu Printer: 0.1.8rc11 so I do get prompted for 2FA authentication and get the new Verify field. When I enter in my password (which I have tested and verified) I get pinged in my email by Bambu labs with the verification code but entering it does not produce an auth token as expected and the field remains unpopulated.

jneilliii, 2024-12-01:
Do you see this same error in your octoprint.log? https://github.com/jneilliii/OctoPrint-BambuPrinter/issues/59#issuecomment-2509573136

singhm29, 2024-12-01:
Feel a bit silly but do I have the right things active to log that?
[image]

I have all three settings active, and I follow the process of entering my login credentials into the plugin. I receive an email from Bambu, enter the code, and click verify, but nothing happens. If I click the verify button again, it triggers Bambu to send another email, but there’s still no update to the authentication token.

I apologize if I’ve missed anything. Please let me know if I need to do something else to capture the correct logging information. I reviewed the logs with those three settings enabled but didn’t see the same output as the other user. It seems like I might have set up the logging incorrectly, as the issue appears identical to theirs.

singhm29, 2024-12-01:
Weirdly the only thing generated is this one log. Not seeing the lines above let me know if I should be doing something more to get them generated other than attempting to log in and to connect to the printer.
[plugin_bambu_printer_serial (2).log](https://github.com/user-attachments/files/17970281/plugin_bambu_printer_serial.2.log)

lordCONAN, 2024-12-01:
Check octoprint.log and search for bambu_printer

jneilliii, 2024-12-01:
Just released a potential fix in https://github.com/jneilliii/OctoPrint-BambuPrinter/releases/tag/0.1.8rc12, but I haven't personally tested it yet. I have merged in the upstream changes for pybambu and adjusted to get it working in local only mode the way I use it. Give it a try and see if your authentication with cloud is working now.

jneilliii, 2024-12-02:
No version will work at this time until I go through the code again. Was hoping for a drop in replacement but appears the ha integration refactored a good bit. I'll be digging in deeper this week in my spare time.

Manmeetmfg, 2024-12-08:
Happy to say I was able to get the authentication token to generate after entering my verification code wiht the latest RC. Thank you @jneilliii for the fix!

singhm29, 2024-12-08:
Was able to generate a verification code and an authentication token with the latest RC. Thank you for the fix @jneilliii!

closed 2024-12-08 by jneilliii

singhm29, 2024-12-11:
Hi, I’m encountering some issues after running more prints. Specifically, I keep getting the error: *“The OctoPrint server is currently not running”* whenever a print is run with the plugin enabled. At first, I thought it might be an issue with OctoPrint itself or another plugin, but I’ve been able to reproduce the problem consistently whenever a print is run.  

If the plugin is connected and idle, everything works fine. Similarly, if OctoPrint is running but disconnected from the printer, it also continues to function without errors.  

Would it help if I posted the output from `Octoprint.plugins.bambu_printer` with serial logging enabled?

jneilliii, 2024-12-11:
it would help to attach octoprint.log with `octoprint.plugins.bambu_printer` logging set to debug potentially. I haven't been running the integration for long periods of time, but never had an issue with it killing the OctoPrint service.

singhm29, 2024-12-12:
Ive attached both. A few things Ive noticed and will try. I am initiating the print through Bambu Studio instead of uploading the gcode to Octoprint and then starting. I am almost certain this was not a problem before 2FA issues but I have noticed that it stays in a state of operational despite the print starting on the Bambu Handy side. I think this makes sense because it doesnt quite know whats going on, it does correctly reflect the tempature changes though. I will try running a print thats been uploaded to Octo to see if behaviour differs. I did also run into filament spool issue midprint in case you notice that but Octoprint crashed before reaching that so not sure it was caught in the log.

[octoprint.log](https://github.com/user-attachments/files/18103653/octoprint.log)
[plugin_bambu_printer_serial.log](https://github.com/user-attachments/files/18103654/plugin_bambu_printer_serial.log)

jneilliii, 2024-12-12:
hmm...looks like it's getting in a file listing loop. can you share the "Elegoogcode.3mf.gcode.3mf" file and possibly the original 3mf file prior to slicing?

singhm29, 2024-12-16:
Apologies for the delay—I wasn’t able to access my printer over the past few days. When I attempted to retrieve the file “Elegoogcode.3mf.gcode.3mf,” it appeared to be corrupted. To address this, I reformatted the SD card (a Sandisk Ultra 16GB) to FAT32 as a precaution.

I also discovered that the “Cache prints to SD card” option wasn’t enabled. After enabling it, the missing progress tracking issue when printing from Bambu Studio was resolved. Now, the progress displays correctly, even when the print wasn’t initiated through OctoPrint. So far, this setup has avoided always crashing the OctoPrint server, even when sending direct prints from Bambu Studio, which used to cause crashes consistently. I have had it go down once after a print with this change however. 

I’m curious if you’re still observing the file listing loop issue. Continuous Print Manager, which worked previously, is also throwing errors and I haven’t found a workaround yet..thinking it might be related.

Lastly I’ve noticed another anomaly: occasionally, the file list doesn’t update properly and remains blank when I connect. Disconnecting and reconnecting OctoPrint usually resolves this on the first attempt, though not always.

I’ve attached updated logs showing a connection to the printer and the “cube.3mf” file sent directly from Bambu Studio. I’m happy to share the G-code and STL files, but they’re currently greyed out as an attachment option here.

[plugin_bambu_printer_serial.log](https://github.com/user-attachments/files/18144399/plugin_bambu_printer_serial.log)
[serial.log](https://github.com/user-attachments/files/18144402/serial.log)

jneilliii, 2024-12-16:
> I also discovered that the “Cache prints to SD card” option wasn’t enabled.

This is definitely a requirement with cloud connected printers, otherwise OctoPrint has no way of identifying. Now that's enabled I'm not seeing the same file list retrieval loop as before. 

> Continuous Print Manager, which worked previously, is also throwing errors and I haven’t found a workaround yet..thinking it might be related.

Not sure, I don't use that plugin but curious if it's what's sending the M400 command on print completion? If not, make sure you don't have anything in OctoPrint's settings under GCODE Scripts. Make all of them empty...

```
2024-12-15 23:08:19,861 <<< N9 M400*46
```