jneilliii/OctoPrint-BambuPrinter #62: [FR]: More than one printer?

metawops, 2024-12-27:
**Is your feature request related to a problem? Please describe.**
I own two BambuLab printers, an A1 mini and an X1C and would love to be able to control both from within OctoPrint.

**Describe the solution you'd like**
Being able to add & configure more than one printer would be great.

**Describe alternatives you've considered**
<none>

**Additional context**
<none>

jneilliii, 2024-12-27:
OctoPrint doesn't support controlling multiple printers at a time, but you can run multiple instances of OctoPrint to control one each instance. I'd suggest looking at the octoprint_deploy script for setting up a second instance.

closed 2024-12-27 by jneilliii

metawops, 2024-12-28:
Ah, I see. Thanks for the hint! 🙏