jneilliii/OctoPrint-BambuPrinter #63: [BUG]: Memory leak in 0.1.8rc14?

metawops, 2024-12-28:
**Describe the Bug**
Every time I activate the plugin – and do nothing in OctoPrint, just let it sit there – the memory amount the LXC uses rises.

**Expected Behavior**
I'd expect that not more and more memory gets used in the container where OctoPrint runs.

**Debug Logs**
[octoprint-logs.zip](https://github.com/user-attachments/files/18266008/octoprint-logs.zip)

**Screenshots**
[image]

**Printer and Plugin Setting Details**

* Printer model?
  * BambuLab X1C
* Is your printer connected to Bambu Cloud?
  * yes 
* Is the plugin configured for local access only?
  * yes

jneilliii, 2024-12-28:
Looks like there may be a related upstream issue on the pybambu module in home assistant. I didn't use cloud mode with my x1c so wonder if it's related to that. If you disconnect the printer in OctoPrint's connection panel does the memory usage go back down?

metawops, 2024-12-29:
I decided to use the A1 mini for the time being with the one running instance of OctoPrint. And I'm currently testing the LAN only mode. The plugin works, I can control the A1 mini, I see the correct filaments from the AMS.
However, memory usage is still growing constantly – regardless of the connection state, so even when disconnected.
I'm happy to help with any further investigation.

metawops, 2024-12-29:
Strange. Sometimes, the memory issue does _not_ happen even when the plugin is active and the connection is active, too.
I have not yet figured out _when_ this happens, under what circumstances. I'll observe ...