jneilliii/OctoPrint-BambuPrinter #65: [BUG]: Printing not possible

metawops, 2024-12-29:
**Describe the Bug**
The plugin seems to work, the A1 mini seems to be connected in LAN only mode. I can control the printer (Home, move x/y/z). The filaments in the AMS are displayed correctly.
However, as soon as I start a print the uploading process to the printer starts but stops almost immediately after < 1 s. 
OctoPrint is not frozen. I can open other dialogs. But nothing gets printed because the file couldn't be sent to the printer.
After a press the Cancel button the State changes to "Cancelling" but stays there with a red error saying "Could not write to serial port".
I have to restart OctoPrint to be able to try again.
The percentage is always different and depends on the file size.
The log files show very strange characters and suggest that the communication is not working properly? 🤷‍♂️ 

**Expected Behavior**
I expected that the print starts.

**Debug Logs**
[octoprint-logs.zip](https://github.com/user-attachments/files/18270217/octoprint-logs.zip)

**Screenshots**
After pressing the Print button this is the state where the process hangs:
[image]

After Cancel button was pressed:
[image]

**Printer and Plugin Setting Details**

* Printer model?
  * A1 mini
* Is your printer connected to Bambu Cloud? 
  * No, it's in LAN only mode 
* Is the  plugin configured for local access only?
  * Yes.

jneilliii, 2024-12-29:
The plugin works differently than most OctoPrint/printer communications. The plugin treats the printer as a virtual sd card, and if you aren't starting a print that is listed as an SD card file the print will not work. Make sure you are uploading files to the SD card side, and starting prints that are stored on SD card. 

[image]

metawops, 2024-12-29:
Ah, thanks, didn't know that. I'm sure some docs are already in the works …? 👍

closed 2024-12-29 by metawops

ados8, 2025-01-12:
I've just discovered this after troubleshooting problem after problem, not easy in the slightest to get Bambu + Octoprint + Obico working. Is there an option to have files auto upload to SD, I haven't found an option or plugin. I'm trying to cut down on the 101 steps to do a print. If I can print from Bambu Studio and it be processed by Octoprint automatically it would be nice. I've been unable to get Obico working. After spending 6+ hours setting up and troubleshooting one of the devs said that Obico won't show a video stream unless using the PI setup and I'm using their docker.

CodeMasterCody3D, 2025-04-18:
I have a script I'll share, it uploads to the SD card automatically then starts the print. I just wish I could implement it in too the repo. Been trying for days with chat gpt but I can just hit upload on orca slicer and it will automatically start the print.

> I've just discovered this after troubleshooting problem after problem, not easy in the slightest to get Bambu + Octoprint + Obico working. Is there an option to have files auto upload to SD, I haven't found an option or plugin. I'm trying to cut down on the 101 steps to do a print. If I can print from Bambu Studio and it be processed by Octoprint automatically it would be nice. I've been unable to get Obico working. After spending 6+ hours setting up and troubleshooting one of the devs said that Obico won't show a video stream unless using the PI setup and I'm using their docker.
> 
> Additionally: I get failure to upload anything to the SD.

jneilliii, 2025-04-18:
The plugin already had the upload to SD card party worked out. How that works from orca is the question.