jneilliii/OctoPrint-BambuPrinter #66: New authentication

jneilliii, 2025-01-16:
Not sure how this will impact the plugin yet.

https://blog.bambulab.com/firmware-update-introducing-new-authorization-control-system-2/

odysseyofmayhem, 2025-01-17:
This commit in BambuStudio caught my attention:
https://github.com/bambulab/BambuStudio/commit/d9ead02cfa1124f875991b59b44b1a60925a432c

Looks like it added the ability to sign or encrypt messages sent to the printer, along with installing a cert on the printer.
All the flats are set to 0 currently, but I think this shows the direction it's going.  At the very least the command messages over MQTT will need to be signed in some way against the cert installed on the printer.

jneilliii, 2025-01-17:
Thanks for the pointers. The direction being taken definitely doesn't bode well for this plugin, and surely not for me to spend any time on future development if they are going to lock it down so much. 

> Also, looks like Bambu have their own "Bambu Farm Manager" which might be another part of this story:
> [wiki.bambulab.com/zh/software/bambu-farm-manager](https://wiki.bambulab.com/zh/software/bambu-farm-manager)

potentially, but a little translation seems to indicate it's only for A1/P1 series devices and not X1 for the "client" connections to the central "server".

biancapascal, 2025-01-20:
The X1 already has a beta running what i did read

poita66, 2025-01-20:
Looks like MQTT will now be available in "Developer Mode", so it's not quite as bad as it was going to be

https://blog.bambulab.com/updates-and-third-party-integration-with-bambu-connect/

I'd love to see this plugin remain actively developed and will try to help out where possible

jneilliii, 2025-01-20:
Yes, the local only developer mode is what will continue to be developed in this plugin. I will not be supporting cloud mode due to the changes, unless Bambu opens a true developer API to the community to use, so expect those changes to be implemented in the near future.

hashoom8, 2025-05-18:
> Yes, the local only developer mode is what will continue to be developed in this plugin. I will not be supporting cloud mode due to the changes, unless Bambu opens a true developer API to the community to use, so expect those changes to be implemented in the near future.

[image]