jneilliii/OctoPrint-BambuPrinter #67: [FR]: Turn on print LED (A1 mini)

Warweedy, 2025-01-21:
**Is your feature request related to a problem? Please describe.**
Add an option in control menu to activate the print LED.

**Describe the solution you'd like**
A button under Fan off for Light On/Off
[image]

**Additional context**
If you are using the MQTT Protocol it works as detailed here:
https://forum.bambulab.com/t/mqtt-for-a1/50033/24

Publish to request topic with body 
`{
"system": {
"command": "ledctrl",
"led_node": "chamber_light",
"led_mode": "on"
}
}`

jneilliii, 2025-01-24:
added in above commit and made available in version 0.1.8rc15. you can change the release channel for the plugin to release candidate in OctoPrint's software update settings and install when prompted.

closed 2025-01-24 by jneilliii

Warweedy, 2025-01-24:
Thank you very much!