jneilliii/OctoPrint-BambuPrinter #69: [FR]:

ulmeck, 2025-01-26:
**Is your feature request related to a problem? Please describe.**
With the recent unpleasaness, I'm trying to remove as much of the Bambu pipeline as possible. So my preferred workflow would be STL -> Orcaslicer -> Sliced .3mf -> Octoprint -> Bambu p1p. 

Right now this mostly works if you select the Octo/Klipper as a 3rd party host and poke it into octoprint. However at this point the file (obviously) is on the octoprint server, not the sd card, and for the life of me I can't see how to transfer it over without downloading and reuploading to SD within octoprint. Apparently Orcaslicer hard codes it as a 'local' upload instead of a 'sdcard' upload, and the simple file manager in Octoprint doesn't seem to have a 'upload local file to SD Card' option, just 'upload new file to SD card'. If you print a local file, octoprint tries (apparently) to interpret the local file as gcode and eats itself.

**Describe the solution you'd like**
When a local file is printed using the Octoprint Bambu plugin, it should first upload the file to the SD card, wait for the upload to finish, and then continue with the print. Alternatively, perhaps it could force all uploads to be treated as uploaded to origin 'sdcard' instead of origin 'local'.

**Additional context**
https://docs.octoprint.org/en/master/api/files.html#upload-file-or-create-folder
[image]

[image]

[image]

[image]

jneilliii, 2025-01-26:
This will probably be changed with the next release of OctoPrint. There are some needed changes in the underlying core to allow this type of workflow and I was already planning on making it work with local files.

CodeMasterCody3D, 2025-04-19:
I opened up a request on OrcaSlicer for this to choose custom POST commands so you could choose the SD card if you wanted. 

For now I just got done making a fork of this repo and adding the addons I made to make orcaslicer and Bambu-octoprint plugin work as a 1 click solution. Took me a few days, well a week..... Well possibly a month 😂 

https://github.com/CodeMasterCody3D/OctoPrint-BambuPrinter/tree/rc

I'll be making a config file soon for the less unfortunate that aren't tech savvy enough to understand any scripts.

For now I am happy and content. Have fun!

jneilliii, 2025-04-19:
Wow, that is some hacked together code. I think I could achieve the same approach much easier and cleaner once I have time to work in the plugin again. OctoPrint 1.12.0 will have the possibility for better integration with Bambu due to changes in the comm layer in OctoPrint but I think 1.11.0 has some changes that will make it possible to use local uploads instead of SD card uploads. Still haven't had time to test.

jneilliii, 2025-04-19:
Sorry, had just woke up when I made that comment and re-read what I wrote and it came across as a bit harsh. I do think I can achieve the upload to SD card in a much cleaner way inside the plugin directly. I'm curious about your camera approach though from a resource perspective. I had heard that opencv processing was used in one of the early versions of OctoPrint but was canned because of performance/processing not being that great. I'd prefer a solution that works for everyone and much like core OctoPrint, webcam streaming is kind of out of scope for what this plugin is intended to do. You can tell even in your code that you are running it as it's own thing, so there's no reason that you can't run that separately. I personally feel like the go2rtc route is a better approach and has worked well for me. 

32 bit OctoPi notes: https://gist.github.com/jneilliii/93b412cb0bbf6b7bfd76f7e10d612f24
64 bit OctoPi notes: https://gist.github.com/jneilliii/d837afbf8a66c716547309f36a598bb0

go2rtc config for bambu (api set to allow my go2rtc plugin to work)
```
api:
  origin: '*'
streams:
  bambu_camera: rtspx://bblp:<access code>@<printer ip address>:322/streaming/live/1
```

jneilliii, 2025-04-20:
I've worked out the automatic upload to printer when sent from the slicer in the above commit. There's some weird caveats though with OctoPrint related to long filenames on SD cards, so if you want the UI to recognize the file sent as printing properly use file names shorter than 8 characters for now.

jneilliii, 2025-05-03:
This is now available in version 0.1.8rc16.

closed 2025-05-03 by jneilliii