jneilliii/OctoPrint-BambuPrinter #71: X1C Camera support

Nicoautoxp, 2025-03-04:
Hi, 
I was wondering how i can add the X1C camera to the octoprint?
Regards, 
Nic

jneilliii, 2025-03-05:
The go2rtc settings for it's config are like this. 

```
streams:
  bambu_camera: rtspx://bblp:<access code>@<ip>:322/streaming/live/1
```

Nicoautoxp, 2025-03-16:
@jneilliii  did you installl go2rtc on the octoprint server?

jneilliii, 2025-03-16:
I've done both actually run go2rtc in the pi and on my HA server. 

For an OctoPi image flash these are my notes for install. 

https://gist.github.com/jneilliii/93b412cb0bbf6b7bfd76f7e10d612f24

CodeMasterCody3D, 2025-04-19:
If anyone needs something quick and easy. My addon to the repo adds camera support for the x1c and 1 click print solution for orcaslicer.

https://github.com/CodeMasterCody3D/OctoPrint-BambuPrinter/tree/rc

skl111, 2025-08-01:
> If anyone needs something quick and easy. My addon to the repo adds camera support for the x1c and 1 click print solution for orcaslicer.
> 
> https://github.com/CodeMasterCody3D/OctoPrint-BambuPrinter/tree/rc

How would I use this until the Octoprint Ecosystem is updated?

BalooDK, 2026-04-25:
> > If anyone needs something quick and easy. My addon to the repo adds camera support for the x1c and 1 click print solution for orcaslicer.
> > https://github.com/CodeMasterCody3D/OctoPrint-BambuPrinter/tree/rc
> 
> How would I use this until the Octoprint Ecosystem is updated?

i found the answer on this link:
https://github.com/CodeMasterCody3D/OctoPrint-BambuPrinter/releases/tag/0.1.8rc15

camera still doesn't work for me though