jneilliii/OctoPrint-BambuPrinter #72: [BUG]: Print jobs load then are immediately canceled.

Ecks32, 2025-03-06:
I've been working with this plugin for a few days.  I wanted to gain some familiarity with it, and do some research, before opening a bug report.  I have started fresh several times as sometimes I can't get it to connect to the printer correctly.  When I can connect everything seems to work other than starting a print.  I put a new SD card in the printer that has only a couple of files loaded.  

Testing procedures.
1. New Octoprint install with Bambu Printer plugin, no other user installed plugins.
2. New SD card in printer properly formatted.  Only files are sliced and uploaded by Orca Slicer.
3. LAN only network.  Isolated from internet.  
4. LAN cable connected to Pi for internet access during install.  The Bambu Printer plugin will not install without internet access.  Cable removed after installation.
5. Verified other Bambu Print plugin features are operational.  Moving/homing, setting temps, ect. I have not tried to get the camera working.
6. Using Auto printer connection settings.  Baud rate it selects is 115200.

I have sliced a test file in Orca Slicer and printed it without Octoprint active.  I start Octoprint and it reads the files on the SD card successfully.  From Octoprint, when I select the test file and click Print it loads the file, then immediately cancels the print.  The screen shot below shows snippets from the Terminal Window.  
[image]

(May not be part of the printing issue)
On every install of Octoprint I've tested, even the ones that don't appear to connect to the printer, I get streams of "Not SD printing" in the terminal, as other users seem to.  With installs that don't look to be connecting to the printer "Not SD printing" is the only messages in the terminal.  On installs that do connect to the printer the "Not SD printing" messages are mixed with the other terminal messages. 

One instance I've made of Octoprint includes the Octodash plugin.  Everything seems to work the same with this plugin installed... I can control everything except starting print files.  However, Octodash doesn't list 3mf files, only gcode files. 

Log files:
[octoprint-logs.zip](https://github.com/user-attachments/files/19111224/octoprint-logs.zip)
[octoprint-systeminfo-20250306083121.zip](https://github.com/user-attachments/files/19111232/octoprint-systeminfo-20250306083121.zip)

**Printer and Plugin Setting Details**

* Printer model? A1
* Is your printer connected to Bambu Cloud?  No
* Is the  plugin configured for local access only? Yes
[image]

Thanks for taking the time to develop this plugin.   I use Octoprint on my other printer, OLD CR10 Hictop I refuse to get rid of.  Being able to use the same control software for both is going to be incredible helpful.

jneilliii, 2025-03-06:
> 4\. LAN cable connected to Pi for internet access during install.  The Bambu Printer plugin will not install without internet access.  Cable removed after installation.

This is a bit of an issue. The pi will need network access to be able to communicate with the printer. Not sure if the network cable was to put this pi into a different VLAN that has internet access or not, but when operating the pi has to be able to communicate with the printer directly over the network. Based on the errors I see in octoprint.log, I think this might be the underlying issue...

> 2025-03-06 08:20:10,207 - pybambu - ERROR - Exception. Type: <class 'OSError'> Args: [Errno 101] Network is unreachable

I see from the logs that you are using the latest 0.1.7 release version, have you tried with the 0.1.8 rc version yet? In software update settings, you'll need to switch the release channel to release candidate for the plugin and upgrade when prompted. 

Also, make sure in your OctoPrint settings, under GCODE Scripts that you have cleared out all the gcode from the available options.

Ecks32, 2025-03-06:
> 4. LAN cable connected to Pi for internet access during install. The Bambu Printer plugin will not install without internet access. Cable removed after installation.

Yes, the lan cable was to connect to a router with internet access.  I set the wireless to connect to my isolated wlan.  
The Octopi had wireless connection to the printer even while connected to the internet with the lan cable.  The shop my printers are located in does not have internet access most of the time so the Octopi won't either. 

I found in my first install of Octopi that once I input my wifi router info then it would never connect to the isolated wlan.  I tried changing the ssid in raspi-config, and the file on the SD card, I think it was wpa_something.txt, and even searched through files on the raspberry pi looking for where it was getting it's connection info.  No matter what I did I could never get it to change networks.  

The only solution I could find was to set the isolated wlan in the rpi manager installation and then connect to the internet with a lan cable to my internet router.  Strange stuff.  

Another issue that is likely related is I keep getting an SD card read error on the printer.  This is why I switched to a new card, in case there was a problem with the old one.  But is seems like whenever I click Print in Octopi the machine gets the SD card read error.  Since I'm not near the machine I only see it if I open Orca Slicer so I wasn't sure if it was related.

Ecks32, 2025-03-06:
And not sure if this means anything but on every install I see a line that says something like "Rais network skipped".  I don't know what that is or if it's related.

Ecks32, 2025-03-06:
SUCCESS!!!  

0.1.8rc.15 Started the print with apparently no problems.  Although, when I first clicked Print, an "Achievement" notice popped up saying I had canceled 10 prints in a day so at first I though it was going to fail again, but it did not.  I'll attach another log in case you want to see what happened.  

The only gcode was the standard end print-cool down code that is preinstalled with Octopi so not sure if that had anything to do with the problems.  The printer cooled down on it's own so didn't need it.  

The added pop up with the printer and AMS settings, as well as the filament loaded in the AMS showing by the Printer Notifications, are a nice touch.  

This plugin seems to be developing nicely.  Great job.

I suppose now I need to see if I can get the camera operational.

jneilliii, 2025-03-06:
Cool, yes. I think your issue was probably resolved with 0.1.8rc6 where I fixed a pathing issue for non-X1 devices.

closed 2025-03-06 by jneilliii