jneilliii/OctoPrint-BambuPrinter #73: Bambu Lab A1 Connection and Printing Issues with BambuPrinter Plugin (v0.1.7)

celsoannes, 2025-03-09:
Hello,

I'm experiencing issues connecting and printing with my Bambu Lab A1 using the OctoPrint-BambuPrinter plugin (version 0.1.7).

Here's a summary of my setup and the problems I'm encountering:

I have successfully installed OctoPrint and the BambuPrinter plugin (version 0.1.7).
I have configured the BambuPrinter plugin settings in OctoPrint Settings as shown in the attached image:

[image]

My Bambu Lab A1 appears to connect successfully, and I can control the printer through OctoPrint.

[image]

[image]

However, when I attempt to start a print, I encounter an error, as shown in the attached image:

[image]

```Recv: T:30.00/ 0.00 B:31.00/ 0.00 @:64
Changing monitoring state from "Operational" to "Starting"
Send: N0 M110 N0*125
Recv: ok
Changing monitoring state from "Starting" to "Printing"
Send: PK\x03\x04\x14\x00\x08\x08\x08\x00\x18siZ\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x13\x00\x00\x00[Content_Types].xml??Kj?0\x10??9???r[(????\x13?\x07\x10??\x16?FB\x1a\x07??\x1d??!?r??>??\x1eW??\x19R?\x01
Send: R]a?0??\x01?\x7f??????x???R?/"?\x03X_?(??2\x13?
Send: ?\x14?Jj?7?????@?\x03PK\x07\x08\x02\x00???\x00\x00\x00?\x01\x00\x00PK\x03\x04\x00\x00\x08\x08\x00\x00\x18siZ\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x14\x00\x00\x00Metadata/plate_1.png?PNG
Send: \x1a
Send: \x00\x00\x00
Send: IHDR\x00\x00\x02\x00\x00\x00\x02\x00\x08\x06\x00\x00\x00?x??\x00\x00\x0e
Send: IDATx\x01??Q\x1cG\x13\x00`I??88??C???,???\x04\x08?\x10\x08?\x0c\x14\x02!(\x04?@\x08?p!\x10???T?=?\x7f?\x07?9?}U[??*?jQmO?t?<{\x06\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00?"??z????????joo???/?,--?z3\x00?\x04????\x1f\x1f\x1f_????_~?e???~????\x7f???O??????????\x02?'??o?9?L?\x02\x7f
Send: ???????Y\x04?}?vtpp0??\x00`?????7o?\x0c#??~?\x13U?j[`T-\x18N?9\x00XP???\x1f??\x05??????_\x17\x0b?^?w??\x01?\x02Z__??\x07?<?????????\x1fFU???\x1b\x04?\x05\x13{????i
Send: ?M??=????G?????El\x1f???\x02?\x02??\x08????\x07?????|\x01\x10O?*H??C\x10c?F\x05\x01?P????????/\x00?*\x02iA?F\x05???/?e\x00(L?????)?O[\x00?[\x04T???4\x00\x14????	??\x16\x04?E??w?,\x02\x00?\x14M?????kE _\x00??"\x16\x19qh????\x07o\x1d\x00\x1eQt?#~#P?\x1b??m	4U\x01?????\x01?\x11???^????\x12?g?\x06?^????\x01\x00?X^?x????g)??\x7f???g?E@}\x0b \x16\x00?_?\x1emmm?\x05\x00??\x10G?FP?\x16?????"?i\x01\x10?\x0c\x0b\x00\x00xd??????L??\x1f\x17??-\x02?2?<?W\x7f????\x16\x00\x00?\x08\x0e\x0e\x0e>?\x07??
Send: ?M\x0b?z???\x01??7?M?\x1b\x1b\x1b\x0e\x06\x02??\x14??i??m\x17\x00?d?Q??????????\x04\x00<???o?i\x7f?\x04???\x00b?0????????s\x00\x00?????g?X??\x7fS???(????????v\x10?\x01?!?\x7f???I\x1d??\x13??)??????????n????\x1f?z?s?	\x00x\x00??_??M7??\x1a??5??l?^??\x05????h????W_\x1d?i\x00?\x03???gUP\x1e?%?q??????\x1a?R?O?}\x11??\x1e\x7f\x04?????????\x0f\x00\x0f?i??)?O??>?W?????_?z\x15??WQ?\x17?\x01?\x01E??????Oz?n?KA????\x02??/?WWW/???????7?\x04\x00???K?m?\x7f?????S?\x0e??????/:???\x0f\x00?lZ?\x7f???u?G???~d?
Send: \x15??E??~\x7fS??????_???	?\x00P?8?'/??\x12??\x7f*??\x18_d??^?\x1aV{???,\x00\x14(J?U?>lS?o??????p??\x0f\x00
Send: ???yQ\x05?y??\x08?????\x00P??????>???o????j1??\x0f\x00Ju_%???W\x0b	Y?\x00?,f?#x?-?7e?qN??\x1f\x00
Send: \x16Y\x7fU??j?????\\x7f??wppp??\x1f\x00
Send: ???m[?O\x17???E???
Send: \x00???\x7f\x15??g-??\x1b?K?~??\x05????????o???\x07\x00\x0b??\x17\x1bU?~y??o??t:??,\x00\x14??%>???4?\x01?????>?\x1a??\x12\x1f?~\x00P???????\x1a?\?\x03\x00?[^^>m[??\x1b?R?\x1a?\?\x03\x00????}6??z?so\x15\x00
Send: \x15%??_\x7fv?O?\x13?"??\x1b?\?\x03\x00????????o??\x1b???\x03@?b???%>????\x1b?\?\x03\x00\x05?g?\x7fR???~\x00(\???i?~.?\x01??E?\x7fttt5O?_???F???\x00@??o?\x7f=???\x07\x00\x0bh?K|?s??F???\x00P??????t?\x7fj????R?\x07??mnn^?F??%?z????\x00\x14.???l\x7f??>?~\x00?@?\?3K?_???aA?,\x00\x14j????F???\x00????\TI?u????6(N??????>\x1a???\x01?=?t:???\x1f?og?\x18]V?F\x15Q?\x10??o?\x1dj?\x03?\x1f??W???
Send: ???+?_?3.?O????\x03???[??I??wm?K??lmm}R?\x02???g???~\x00(\d?U??j??\x7f,\x18?~\x1a?\x00`A????6%?z?_o?3?\x0f\x00????}5??l??T\x00(?]f??9??%>J?\x00P?Yg?'e?.?\x01?\x05\x11?U?~\x19\x19?]???F?N?s??\x02@????7???\x1b???\x01?`???????F?%\x7f\x00(\d???|5?\x01?\x13???|:K?\x7fR?_:?/J?q???~\x00(X*??s?_??/.???z??*\x00\x14??8?4????\+?\x03@??-?k?\x03?\x05U/?O????5?E?7?U??\x13o\x15\x00
Send: \x15%?*p_?[?7???\x07????7??%>+++\x17?*\x00\x14?m?\x7fR???~\x00(\?????>\x1a?\x00??????D?I?~.?\x01??E???y?\x7frrr?h?\x03?'???T\x19?MS??\x17\x00?\x1a?R??\x0f\x00\x16D????\x02?m?O?~\?oj?KY?F?\x00X\x10?????\x1e?S?4??\x0f\x00\x16L\x04???J?M??x\x1f\x00,???\x1f\x17????y???\x1f\x00\x16T\x04?|?o????\x1f\x00\x16X\x1e?'??S?w???\x01`\x01???n????c????G?7?\x0f\x00\x0bhgg?1?????????`???t?
Send: \x02???\x14??\x0e?h?????z??\x1e\x00,??t\x7f^??????\x1f??\x11???\x07\x00\x0b???O"???S????G??}~\x00XpUV\x7f??_\x7f??\x7f\x7f\x1b?????i?_?\x1f\x00<\x11?N??????????c?_?\x07?'(\x16\x00\x11????\x7f??\x17
Send: ~???\x02?\x00<UU?\x7fQe????\x1c\x0c\x06W?~?\?\x07\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00Xx?\x03c??
Send: ??\x1dJ\x00\x00\x00\x00IEND?B`?PK\x07\x08??F\x0e\x00\x00F\x0e\x00\x00PK\x03\x04\x00\x00\x08\x08\x00\x00\x18siZ\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x1a\x00\x00\x00Metadata/plate_1_small.png?PNG
Send: \x1a
Send: \x00\x00\x00
Send: IHDR\x00\x00\x00?\x00\x00\x00?\x08\x06\x00\x00\x00?>a?\x00\x00\x03^IDATx\x01??OHTQ\x14???\x19}??8????\x1fG??A\x10\x18A?L???\x16-???:\x08$?\x16A\x11????+?m??M\x11BE?h'\x12A?2?2\x0b?????W?J???\x07\x06u?F???s?	?\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00?_????d~\x12???F????????????w??\x04HKKKatt????
Send: ??D?r??+J??\x7f{???\x16?\x14????:
[... cut ...]
Could not write anything to the serial port in 5 tries, something appears to be wrong with the printer communication
Recv: Not SD printing
Connection closed, closing down monitor```

If I try to cancel the print, it also gives an error, as shown in the attached image.

<img width="157" alt="Image" src="https://github.com/user-attachments/assets/630bb0db-6d1c-4980-8177-9dc16f912c68" />

I'm unsure what I'm doing wrong. Could someone please provide guidance on how to resolve these connection and printing issues?

Any help would be greatly appreciated.

Thank you.

celsoannes, 2025-03-09:
I updated to Version 0.1.8rc15, and only thing I get more is to select the color of the fillament, but still not working:

```Recv: Not SD printing
Could not write anything to the serial port in 5 tries, something appears to be wrong with the printer communication
Recv: Not SD printing
Connection closed, closing down monitor
```

jneilliii, 2025-03-09:
The plugin doesn't work with files uploaded to OctoPrint's local storage. You upload to SD card, or send direct to printer from slicer. Also, remove any settings you have in OctoPrint's gcode scripts section.