jneilliii/OctoPrint-BambuPrinter #75: [FR]: I now have 1 click printing

CodeMasterCody3D, 2025-04-19:
**Is your feature request related to a problem? Please describe.**

I was tired of having to jump through hoops, so I made a python script, and a bash script, the bash script is for our x1c camera,  the python uses the octoprint API to hook on to the pybambu that you use in this repo.
**Describe the solution you'd like**

My simple solution works wonders, I worked hard on this and I am a noob, I used chat gpt and lots of testing.
**Describe alternatives you've considered**

I would love for you to implement this python code into your repo because I tried it using chatgot and I can't get anywhere modifying the repo. I'm not very good at python.
I can think of good ideas and  if I surround myself with enough smart people I'll be able to fulfill my dreams. 
**Additional context**

[image]

CodeMasterCody3D, 2025-04-19:
Also with the script it uses the correct time for the print!!!!

CodeMasterCody3D, 2025-04-19:
https://github.com/CodeMasterCody3D/OctoPrint-BambuPrinter/tree/rc

I'd love to work on this with you to make this the greatest Bambu 3rd party host for end users that don't have a farm. Since bambu-farm is available, but can't send prints directly to it... If you have a farm you don't need to slice files all the time so, when I realized there was basically no 3rd party good way to send prints wirelessly I had to jump on this and help, octoprint is a great platform for this!!

CodeMasterCody3D, 2025-04-19:
Oh and btw 8 is my lucky number. Lolololol I just noticed I have 8% battery in the screen shot lolol

CodeMasterCody3D, 2025-04-21:
Man if you could implement this better than me I'd be so happy because man. I been working on it like a maniac.

I'm glad you noticed the 8.3 SD card short name thing. That was something that took a long time to figure out the first time. 

I just got done doing another round of working on this and got the glitch figured out where if you cancel at certain times it would not print the next print,.

I found out you can either disconnect reconnect serial or just select a random file.

Chat gpt said it was because if the octoprint API sees the file as selected in those times it will try and start it but it technically isn't selected. So I set an if statement where if it tries to keep uploading my file (which means print hasn't started) it will disconnect reconnect and it will then try and upload again and start and it successfully starts printing after that.

Using the octoprint API post commands has been a life saver. 

If you can add check box variable somewhere, where we can automatically start the print that would be great. Since orca slicer only tries to start a print with post commands only on local storage of the octoprint and not the SD card.

My implementation is wack I bet to a real programmer.

Newest version of my auto upload with the weird fixes and stuff it's probably over doing it but it works and I hope you can take from this code and make it better. Feel free to use my logo from my repo. You don't have to give me credit.

```sh
#!/usr/bin/env python3
import os
import yaml
from time import sleep, time
import subprocess
import requests
from pathlib import Path
from watchdog.observers import Observer
from watchdog.events import FileSystemEventHandler
from datetime import datetime

def get_octoprint_setting(keys, fallback=None):
    # Use Docker-specific path or fallback to default
    possible_paths = [
        "/octoprint/octoprint/config.yaml",  # Your Docker mount
        os.path.expanduser("~/.octoprint/config.yaml")  # Default OctoPrint path
    ]
    
    for config_path in possible_paths:
        if os.path.exists(config_path):
            try:
                with open(config_path, "r") as f:
                    config = yaml.safe_load(f)
                value = config
                for key in keys:
                    value = value.get(key)
                    if value is None:
                        return fallback
                return value
            except Exception as e:
                print(f"Error reading config at {config_path}: {e}")
                return fallback
    print("No config file found.")
    return fallback

# ─── CONFIGURATION ─────────────────────────────────────────────────────────────
WATCH_DIR = get_octoprint_setting(["folder", "uploads"], "/octoprint/octoprint/uploads")
API_KEY = get_octoprint_setting(["api", "key"], "")
API_URL_SD     = "http://localhost:80/api/files/sdcard"
API_JOB        = "http://localhost:80/api/job"
LOG_FILE       = os.path.expanduser("~/print_watcher.log")
API_CONNECTION = "http://localhost:80/api/connection"  # Update this URL if your OctoPrint API uses a different host or port
VALID_EXTS     = {".3mf", ".gcode", ".stl"}     # Accepted file types
STABILIZE_SEC  =  2    # seconds between size checks
STABILIZE_CNT  =  3    # consecutive identical-size checks
# ────────────────────────────────────────────────────────────────────────────────

def log_message(msg: str):
    ts = datetime.now().strftime("%Y-%m-%d %H:%M:%S")
    with open(LOG_FILE, "a") as f:
        f.write(f"{ts}: {msg}\n")
[... 361 lines cut ...]
        """Handle file creation events."""
        log_message(f"Handling created event for file: {event.src_path}")
        self.process_file(event)

    def on_modified(self, event):
        """Handle file modification events."""
        log_message(f"Handling modified event for file: {event.src_path}")
        self.process_file(event)
        # Optionally remove local copy:
        # fp.unlink()
        # log_message(f"Removed local: {full_name}")

if __name__ == "__main__":
    log_message(f"Watcher starting on {WATCH_DIR}")
    obs = Observer()
    obs.schedule(PrintWatcher(), WATCH_DIR, recursive=False)
    obs.start()
    try:
        while True:
            sleep(1)
    except KeyboardInterrupt:
        obs.stop()
    obs.join()
    log_message("Watcher stopped.")
```

CodeMasterCody3D, 2025-04-21:
curl -s -X POST "http://192.168.2.78:80/api/files/sdcard/current" \
     -H "X-Api-Key: YOUR_API_KEY" \
     -H "Content-Type: application/json" \
     -d '{"command":"unselect"}'

curl -s -X POST "http://192.168.2.78:80/api/files/local/current" \
     -H "X-Api-Key: YOUR_API_KEY" \
     -H "Content-Type: application/json" \
     -d '{"command":"unselect"}'

curl -s -X GET "http://192.168.2.78:80/api/job" \
     -H "X-Api-Key: YOUR_API_KEY"

curl -s -X POST "http://192.168.2.78:80/api/job" \
     -H "X-Api-Key: YOUR_API_KEY" \
     -H "Content-Type: application/json" \
     -d '{"command": "cancel"}'

curl -s -X GET "http://192.168.2.78:80/api/files/local" \
     -H "X-Api-Key: YOUR_API_KEY"

curl -s -X GET "http://192.168.2.78:80/api/files/sdcard" \
     -H "X-Api-Key: YOUR_API_KEY"

curl -s -X POST "http://192.168.2.78:80/api/files/local" \
     -H "X-Api-Key: YOUR_API_KEY" \
     -F "file=@/path/to/your/file.gcode"

curl -s -X DELETE "http://192.168.2.78:80/api/files/sdcard/<path-to-file>" \
     -H "X-Api-Key: YOUR_API_KEY"

jneilliii, 2025-04-21:
> If you can add check box variable somewhere, where we can automatically start the print that would be great. Since orca slicer only tries to start a print with post commands only on local storage of the octoprint and not the SD card.

With my changes, if the upload and print button is pressed in orca it will auto select and start the print. Otherwise it just uploads any file sent to local storage in OctoPrint to the printer.

CodeMasterCody3D, 2025-04-21:
> With my changes, if the upload and print button is pressed in orca it will auto select and start the print. Otherwise it just uploads any file sent to local storage in OctoPrint to the printer.

OMG thank you!!! I hated having to start my scripts manually during all this testing. I was gonna make a service and have it boot up but this is way better. Thank you!! Please consider using my logo from the forked repo I did of yours. I'll be spreading this around on reddit.

jneilliii, 2025-04-21:
Once I figure out the file naming issue I'll release with the next RC version. I vaguely remember this being a limitation of OctoPrint's core when working with SD cards. When the plugin loads the file list from the printer via ftp it adds the 8.3 filename to handle this, probably just need to add the code to update the file list after upload, then use the shortened name from the list for the select command.

CodeMasterCody3D, 2025-04-22:
> Once I figure out the file naming issue I'll release with the next RC version. I vaguely remember this being a limitation of OctoPrint's core when working with SD cards. When the plugin loads the [...]

I found this information!!!

https://gist.github.com/foosel/cadc38123687a46816a1d3e1d160d48a

```sh
# -*- coding: utf-8 -*-
from __future__ import absolute_import, division, print_function, unicode_literals

import octoprint.plugin
import octoprint.events
import octoprint.util.comm

class M33StormPlugin(octoprint.plugin.EventHandlerPlugin):
	def __init__(self):
		self._active = False

	def on_event(self, event, payload):
		if event != octoprint.events.Events.UPDATED_FILES:
			return

		if payload.get("type") != "printables":
			return

		if self._active:
			return

		sd_files = self._printer.get_sd_files()
		commands = list(map(lambda x: "M33 /{}".format(x["name"]),
		                    filter(lambda x: x["display"] == None,
		                           sd_files)))

		if commands:
			self._active = True
			def process():
				self._active = False
			marker = octoprint.util.comm.SendQueueMarker(process)
			commands.append(marker)
			self._printer.commands(commands)

__plugin_name__ = "M33 Storm"
__plugin_description__ = "Requests long names for files on printer's SD card via M33"
__plugin_author__ = "Gina Häußge"
__plugin_implementation__ = M33StormPlugin()
```

Also I found that adding sleep(5) to

```sh
@gcode_executor.register("M524")
    def _cancel_print(self):
        self._current_state.cancel_print()
        time.sleep(5)
        self.remove_project_selection()
        return True
```
Will let me seamlessly cancel prints and restart them.

I also added logic to the file upload you implemented, so it renames the SD card file names incrementally if the file being uploaded matches only the 8.3 name, if it matches both it replaces the 8.3 file name.  

```sh
def upload_to_sd(

            self,

            printer,

            filename,
            path,
            sd_upload_started,
            sd_upload_succeeded,
            sd_upload_failed,
            *args,
            **kwargs,
        ):
        self._logger.debug(f"Starting upload from {filename} to {filename}")
        sd_upload_started(filename, filename)

        def process():
            with measure_elapsed() as get_elapsed:
                try:
                    with self._bambu_file_system.get_ftps_client() as ftp:
                        existing_files = ftp.list_files()

                        # Get base name and extension
                        base_name, ext = os.path.splitext(filename)
                        prefix = base_name[:6].lower()

                        # Look for similar files that might conflict
                        existing_filenames = [os.path.basename(f["name"]).lower() for f in existing_files]
                        similar_files = [f for f in existing_filenames if f.startswith(prefix) and f.endswith(ext.lower())]

                        # If exact name exists, delete it (safe overwrite)
                        file_to_delete = next((Path(f["name"]) for f in existing_files if os.path.basename(f["name"]) == filename), None)
                        if file_to_delete:
                            self._logger.info(f"File '{filename}' already exists on SD card. Deleting '{file_to_delete}' before upload.")
                            self._bambu_file_system.delete_file(file_to_delete)

                        # Handle short name collision (e.g., cube_p~1.3mf)
                        if filename.lower() in existing_filenames:
                            new_index = 1
                            while True:
                                new_filename = f"{prefix}_{new_index}{ext}"
                                if new_filename not in existing_filenames:
                                    filename = new_filename
                                    break
                                new_index += 1
                            self._logger.info(f"Filename collision detected. Uploading as '{filename}' instead.")

                        # Upload file
                        if ftp.upload_file(path, filename):
                            sd_upload_succeeded(filename, filename, get_elapsed())
                            # Refresh file list
                            self.refresh_file_list()  # Trigger file list refresh after upload
                        else:
                            raise Exception("upload failed")
                except Exception as e:
                    sd_upload_failed(filename, filename, get_elapsed())
                    self._logger.exception("Upload failed with exception", exc_info=e)

        thread = threading.Thread(target=process)
        thread.daemon = True
        thread.start()
        return filename

    def refresh_file_list(self):
        """Method to trigger a refresh of the file list."""
        self._logger.debug("Refreshing file list.")
        # Assuming you have access to OctoPrint API to trigger refresh, you can use:
        self._printer.commands("M20")  # This command triggers a refresh of the SD card file list
```

Works great, but it's not refreshing SD list automatically. 
I'm new to all this and never learned anything professionally and or went to school for any of this. I am using chatgpt And I ask a lot of questions and I use logical thinking to come up with ways to program things. I like to modify, idk if I will ever be able to build something from scratch like this plugin but I love to test things and figure things out..

Do you know why it's not refreshing?

jneilliii, 2025-04-22:
> I found this information!!!

This M33 implementation isn't really going to help at all because our file list has to be generated over ftp once the upload is complete. It won't be difficult to figure out, just need your patience and some of my time to do it. 

> Also I found that adding sleep(5) to
> 
> @gcode_executor.register("M524")
>     def _cancel_print(self):
>         self._current_state.cancel_print()
>         time.sleep(5)
>         self.remove_project_selection()
>         return True
> Will let me seamlessly cancel prints and restart them.

Looks like you also added remove_project_selection()? Current code doesn't have that there with the expectation that it happens at the beginning of the printing state change. 

https://github.com/jneilliii/OctoPrint-BambuPrinter/blob/c1e96b230618d55862cb34cfa5661b8d12f1ad96/octoprint_bambu_printer/printer/states/printing_state.py#L31

and gets called here: https://github.com/jneilliii/OctoPrint-BambuPrinter/blob/master/octoprint_bambu_printer/printer/bambu_virtual_printer.py#L674

```
    @gcode_executor.register("M524")
    def _cancel_print(self):
        self._current_state.cancel_print()
        return True
```

> never learned anything professionally and or went to school for any of this.

Me either, writing plugins is how I got into it, without the help of any of the LLMs.

CodeMasterCody3D, 2025-04-23:
Ahhh, only thing I could think of is, at the begining of the printing stage the remove project section doesn't have enough time to  remove it before it tries to start again, I assume. Idk 

Idk I did a lot of testing and that was the only way I could, cancel via octoprint, then start the same name print again via orca slicer.

If I don't have 

```sh
sleep(5)
self.remove_project_selection()
```

On the cancel section with the sleep 5, I can't seamlessly send another print and it start it again if I don't have that in there.

jneilliii, 2025-04-26:
The above commit should finalize the 1-click printing from Orca now, it's able to recognize the printing file now after switching to using the 8.3 format for file selection. If you want to test you can use this URL in Plugin Manager > Get More > ...from URL.

```
https://github.com/jneilliii/OctoPrint-BambuPrinter/archive/refs/heads/rc.zip
```

I have temporarily taken out timelapse file listing until I figure out what's causing log spamming on generating the 8.3 format names.

CodeMasterCody3D, 2025-04-26:
I just tried it. Sam glitch, it won't switch back.

If I select the SD card file
[image]

Then say if someone accidentally selects the local file 
[image]

It won't switch back
[image]

CodeMasterCody3D, 2025-04-26:
> The above commit should finalize the 1-click printing from Orca now, it's able to recognize the printing file now after switching to using the 8.3 format for file selection. If you want to test you [...]

Wait I'm a little confused. 

What did you fix?

CodeMasterCody3D, 2025-04-26:
> The above commit should finalize the 1-click printing from Orca now, it's able to recognize the printing file now after switching to using the 8.3 format for file selection. If you want to test you [...]

I did notice the file on the SD card being replaced. 

i thinking the Bambu virtual printer is fixed so that it clears the selection that octoprint is holding on too, it's the name specifically. 

Because if I choose the same SD card name after switching to local it will not switch back, but I can switch to another named SD card file. 

I also noticed if I disconnect the serial then reconnect it clears the selection and I can select the SD card file I was trying to select but if I put it back in the glitched loop it will get stuck on a local file and will not select the same named SD card file, it's holding on the the display name and not checking against the SD card 8.3 short name to the long file name. I assume. Idk

jneilliii, 2025-04-26:
https://github.com/user-attachments/assets/23cd8823-d164-4ce9-be6f-86604d14cf61

jneilliii, 2025-04-26:
> I also noticed if I disconnect the serial then reconnect it clears the selection and I can select the SD card file I was trying to select but if I put it back in the glitched loop it will get stuck [...]

I think this is a separate issue, if the file name of the 1-click file is different than the one that was previously sent/printed it should work like in that video above. I am seeing the same thing as you though with the not being able to send the same file name, selection weirdness.

jneilliii, 2025-04-26:
I may have tackled the reprinting issue in the commit above. You can reinstall with the same url as before to try it out and verify.

jneilliii, 2025-05-03:
New release 0.1.8rc16 with the latest changes including 1 click printing. Please test it out and let me know.

jneilliii, 2025-05-03:
If you're still having issue with print starting please open separate issue with debug logging enabled and share those logs.

closed 2025-05-03 by jneilliii