jneilliii/OctoPrint-BambuPrinter #79: Remaining print time, A1.

Ecks32, 2025-05-10:
I'm just wondering if the plugin is supposed to be getting the print time left.  You can see in the attached pic that it looks as if it tries to report the remaining print time but it's off by a few... months.  I'm not sure if it's something that's been implemented or not.

This is for my A1.
Plugin Version 0.1.8rc15
Using LAN only mode, the printer does not have internet access. 
Print time genius is installed. 

The web interface shows a wild print time remaining and I don't see any reference to the current layer.
The Raspberry Pi screen doesn't show any print times or layers.

I did just notice that I have an old version of Octoprint.  When the current print is complete I'll have to go get it and give it internet access to update.

Thanks
[image]

jneilliii, 2025-05-10:
Currently the plugin does not do anything related to time estimation. What you are seeing is the result of OctoPrint core calculating the time remaining from bytes reporting from the plugin and time elapsed since start. If you aren't seeing the size go up from 1 byte them that's why it's not even close.

Ecks32, 2025-05-10:
I thought that may be the case but I didn't really know for sure.  Since Orca Slicer puts the time estimate in the filename, and Octoprint keeps track of how long the print has been going, it's easy to get an idea of remaining print time.  

This plugin has made using the A1 in LAN mode a viable option.  I had been using it in LAN mode but it was a discombobulated process of switching back and forth between four programs.  This plugin has brought that down to the slicer and Octoprint, just like with my other printers.  I appreciate all your hard work.  

Thanks
X32

jneilliii, 2025-05-10:
Plugin is obviously still a little buggy with some things, and not having an open standard to deal with Bambu it has been a challenge.

closed 2025-05-10 by jneilliii