jneilliii/OctoPrint-BambuPrinter #84: [BUG]: Print jobs immediately cancel after starting

mitsake, 2025-06-16:
**Describe the Bug**
When a print is selected from the left-hand file menu, the print job starts and immediately cancels. The behaviour looks very similar to that described in #72, though I'm running version 0.1.8rc16. Printing from Bambu Studio works as normal and temperature etc. appears to be monitored fine through Octoprint.

**Debug Logs**
I've cut these to a section surrounding one of the start/cancel events - please let me know if I've removed too much.

* [plugin_bambu_printer_serial.log](https://github.com/user-attachments/files/20758100/plugin_bambu_printer_serial.log)
* [octoprint.log](https://github.com/user-attachments/files/20758102/octoprint.log)

**Screenshots**

[image]

**Printer and Plugin Setting Details**

* Printer model?
  * P1S (firmware 01.07.00.00 - I think the next step is probably to update and turn on developer mode)
* Is your printer connected to Bambu Cloud? 
  * No
* Is the plugin configured for local access only?
  * Yes

jneilliii, 2025-06-22:
would recommend clearing the gcode scripts from OctoPrint's settings. 

looking at the log the file that was started was called "hex.stl -1 + hex.stl -1 + hex.stl -1 + hex.stl -1 + hex.stl -1 + hex.stl -1 + hex.stl -1 + hex.st....gcode.3mf"? I'm confused about all the periods at the end and wonder if that might be causing this. have seen some 3mf project files downloaded and printed directly can cause weird behavior if the project name didn't match the file name.

mitsake, 2025-06-28:
> would recommend clearing the gcode scripts from OctoPrint's settings.

Apologies for the delay in getting back to you - this advice fixed the problem completely. The only script I had to remove was the default [`afterPrintCancelled`](https://docs.octoprint.org/en/master/features/gcode_scripts.html#bundled-scripts). I imagine there's a command in there that my P1S doesn't like, but I haven't looked into it any further.

> looking at the log the file that was started was called "hex.stl -1 + hex.stl -1 + hex.stl -1 + hex.stl -1 + hex.stl -1 + hex.stl -1 + hex.stl -1 + hex.st....gcode.3mf"? I'm confused about all the [...]

This is due to how BambuSlicer names files when you haven't saved a project - it just uses the name of the current model, and if you clone that model multiple times (which I had because I needed some tokens for a teaching activity), it duplicates those in the filename too. That was an unusually large one, so I'll keep an eye on it when printing in future.

In case they're useful, here are the logs from a recent successful print using the plugin:

- [octoprint.log](https://github.com/user-attachments/files/20958449/octoprint.log)
- [plugin_bambu_printer_serial.log](https://github.com/user-attachments/files/20958450/plugin_bambu_printer_serial.log)

I'll close this for now - please let me know if there's anything I can help with. I'm not really a programmer, but could possibly help out with documentation?