jneilliii/OctoPrint-BambuPrinter #87: Plugin 0.1.8rc16, A1 Connection Issues

alexjn1984, 2025-09-01:
Hey there, 

I really only use OctoPrint for timelapse and status updates on the printer. I know about all of Bambu's firmware junk, and I can live with it. But now I'm not sure what's going on.

I've set the A1 to LAN only, and dev mode enabled.

Octoprint sometimes will connect, often times not. WHen it does, its not showing status properly of the device, nor loading the SD card contents properly (new behavior).

Its also doing a ton of not SD printing, even though I have a print going, which was started by Bambu studio, also through LAN mode.
Attached is my debug log, although theres not much info in there from what I can see.

I have a print going right now, once it finishes I'm going to try rebooting the printer, its the only thing I haven't rebooted at this point.

Thank you!

[plugin_bambu_printer_serial(2).log](https://github.com/user-attachments/files/22070868/plugin_bambu_printer_serial.2.log)

jneilliii, 2025-09-01:
Yeah, the log doesn't really show much for debugging purposes. It does show the plugin was able to pull a file list. Any errors in octoprint.log? Did you have debug mode enabled in OctoPrint's logging section of settings for octopring.plugins.bambu_printer? Restarting the printer and then disconnect/connect from OctoPrint might help.

alexjn1984, 2025-09-01:
seeing this a bunch
2025-09-01 01:28:30,320 - octoprint_bambu_printer.printer.pybambu - WARNING - On Disconnect: Printer disconnected with error code: 7

Octoprint is running on a pi connected via ethernet. The A1 obviously wifi.
-68 dBm
TX Retries
0.0%
Standard
WiFi 4
MIMO Configuration
1x1
Rx Rate
150 Mbps
Tx Rate
150 Mbps

[octoprint(2).log](https://github.com/user-attachments/files/22071580/octoprint.2.log)

[plugin_bambu_printer_serial(5).log](https://github.com/user-attachments/files/22071588/plugin_bambu_printer_serial.5.log)

alexjn1984, 2025-09-02:
Yeah no dice. I know it worked before so I'm blaming it on Bambu.

Thank youf or the help!

closed 2025-09-02 by alexjn1984

alexjn1984, 2025-09-06:
Random question,

When first communicating, does it send any commands that interact with the camera module to confirm its running? Apparently my module is bad and they are sending a replacement. Just a thought.

alexjn1984, 2025-09-07:
Once I get the replacement I'll give it a try and let you know

jneilliii, 2025-09-07:
The alternative is to manually override the config passed to the client initialization. Find the file bambu_virtual_printer.py under site-packages and in this code block: 

https://github.com/jneilliii/OctoPrint-BambuPrinter/blob/rc/octoprint_bambu_printer/printer/bambu_virtual_printer.py#L261-L276

Make it this (add the enable_camera property)

```py
        bambu_client = BambuClient(
            {"device_type": self._settings.get(["device_type"]),
            "serial": self._settings.get(["serial"]),
            "host": self._settings.get(["host"]),
            "username": (
                "bblp"
                if self._settings.get_boolean(["local_mqtt"])
                else self._settings.get(["username"])
            ),
            "access_code": self._settings.get(["access_code"]),
            "local_mqtt": self._settings.get_boolean(["local_mqtt"]),
            "region": self._settings.get(["region"]),
            "email": self._settings.get(["email"]),
            "auth_token": self._settings.get(["auth_token"]) if self._settings.get_boolean(["local_mqtt"]) is False else "",
            "enable_camera": False,
             }
        )
```
then restart OctoPrint.